vedderb/bldc #7: safe start bug? when exceeding the pulse width range, full throttle until return to neutral

erwincoumans, 2015-06-08:
reproduction:
choose PPM, enable safe start, duty cycle (for easier test on bench)
set the min pulse width and max pulse width a little bit so that the servo signal exceeds it
now exceed the 'max pulse width' by going full throttle, and then slowly release the throttle.

The signal stays full throttle, until returned to 'neutral' central position.

The workaround is to always make sure the servo never exceeds the min/max pulse width.

vedderb, 2015-06-09:
This bug should only be present while BLDC Tool is connected. The reason for that is that BLDC Tool will prevent the timeout that stops the motor from occurring when the PPM signal is gone, which makes the motor run with the last set speed. When the servo signal comes back, safe start assumes that the motor isn't running and prevents all control commands before the neutral position has been active for 50 iterations.

This bug is now fixed in FW 1.4.

closed 2015-06-09 by vedderb

erwincoumans, 2015-06-09:
thanks for the fix!