vedderb/bldc #17: Tachometer stops working after applying brake, when motor was running at high duty.

pabloxid, 2016-04-09:
Steps to reproduce this error:

1) run BLDC tool, select "realtime data" tab, check "activate sampling"
2) send some kind of "high duty" command to the motor (e.g. duty 0.5, rpm 4000, current 3.0A, etc.)
3) then press "brake" / "full brake" button (or send duty 0.0) 
4) now move motor shaft by hand
5) repeat from point 2)

You may notice after repeating this several times, some of them tachometer don't change with manual shaft movement, it remains "frozen" until you quit brake state.

rewolff, 2016-04-16:
It could be that the measuring of the rotor state is difficult when "full brake" is on. So I wouldn't be suprised it it cannot measure the rotor state in that situation. Could it be that you're getting tired after a a few tries so that the induced voltages are less and that the measured voltages end up below the detection threshold?
Why is this a problem for you?

pabloxid, 2016-04-16:
Yeah, I thought this might be the cause, however there are two important points:

1) failure occurs only sometimes. That is, the system is actually capable to detect the state of the rotor when "full brake".

2) this shouldn't be a surprise, because in fact I'm using hall sensors. Maybe the problem is that, occasionally the system is put into sensorless mode and "forget" that it has hall sensors (In fact, it switches to sensorless mode at high duty cycles, don't it?).

Why is this important to me? Because I'm using VESC in a positioning system, so the tachometer can help me determine if there was any fault.

pabloxid, 2016-04-17:
Definitely the problem is related to sensored/sensorless switch under "hybrid" mode, which is what I've been using. If I put VESC into full "sensored" mode, tachometer works always.

pabloxid, 2016-04-17:
I think when braking, a call to update_sensor_mode () is missing somewhere in the code. Can anyone confirm this, and tell me where that call should be placed?

Thank you.

vedderb, 2016-04-18:
I can have a look at this soon. Sensored mode in the BLDC code has some issues in general and I haven't tested it much, so it needs some improvements. Could you give sensored FOC a try?

pabloxid, 2016-04-18:
I could not get my motor works in FOC mode. Maybe I didn't understand the procedure for adjusting the parameters, but for reasons of time, I needed to finish this project and had already achieved some results in BLDC mode, so I kept in BLDC mode. This issue is for "hybrid" BLDC mode, full "sensored" mode works well.