vedderb/bldc #30: Explicit liense?

mkschreder, 2017-03-07:
I see most of the source code has GPLv3 in the headers, but nothing about overall license is mentioned in the readme. 

Could you ensure that all files in the project comply with one license and add a notice about the overall license to the readme?

PaulMakesThings, 2017-03-08:
Probably not a usual reason that would apply to a lot of people.

I'd been looking for a platform to prototype a watch that could show deaf
people what sounds were around them and what people were saying, and maybe
what direction it was coming from but that would require two mics. But most
smartwatches lack the power and few actually run an OS that I could develop
on.

So when I saw a smart watch with a processor that could run my app on
android I got all excited, then later realized that though it tics all the
really hard boxes it lacks a microphone. Which makes sense since you made
it for gaming.

It looks like a great product, though, I have nothing against the design or
how you're making it.

Thanks!

vedderb, 2019-02-19:
It is now in the README

closed 2019-02-19 by vedderb