vedderb/bldc #33: [Feature Request] Add support for Sine-Cosine analog encoders

tlalexander, 2017-05-12:
Apologies if this is not the right place for this, but I would like to add support for Sine-Cosine encoders such as the [KMZ60](http://www.nxp.com/products/automotive-products/sensors/angular-sensors/angle-sensor-with-integrated-amplifier:KMZ60) to the VESC. (Files for my encoder are [here](https://github.com/tlalexander/MotorDriver/tree/master/Angle%20Sensor).)

From what I can gather from a quick examination, I should add a new encoder type to encoder.c, give it its own init function, and return a float angle on encoder_read_deg(). Seems easy enough.

Where do I patch this in to the configuration UI? I've built that too. I suppose that's a different project however...

Anyway, I didn't see a good place to discuss this so perhaps this can serve as a discussion board.

I will be happy testing on my own, but feel free to advise if anyone is interested.

gpzhaome, 2017-05-12:
Hi, 

Not sure I understand your problem correctly. But you can find the encoder configuration (reading from the UI) in conf_general.c . If you want to change the UI, you need to go to bldc-tool project (https://github.com/vedderb/bldc-tool) and change files accordingly.

tlalexander, 2017-05-12:
I've got the option added to the bldc-tool. That was easy.

[image]

I will have to look at how to init the analog pins appropriately and write my encoder routine. I'll save that for another time. Bed time right now.