vedderb/bldc #35: [feature request] Software direction reverse?

l29ah, 2017-07-27:
It would be awesome to be able to switch the "primary" direction of the motor without having to rewire the motor connection (and re-detect its sensors in case of the sensorful operation). Can you advise me on the required code modifications? I'm afraid to burn the motor and VESC MOSFETs or do lots of extra work (even `mcpwm.c` looks quite complex, and it seems like I need modify much more than `set_next_comm_step()`).

DieBieEngineering, 2017-07-27:
Software direction change is already implemented in the latest (not yet publicly available ) version of the VESC firmware. Which if I understood correctly will be released in a couple of weeks. I know its not an answer to your question, I just wanted to point out that there is a significant update coming!

l29ah, 2017-07-27:
Oh, good to know, thank you!

vedderb, 2019-02-19:
Implemented a while ago, closing.

closed 2019-02-19 by vedderb