vedderb/bldc #48: DC Switching Frequency

Hug0, 2017-12-20:
In DC mode, calculations relative to motor current are not consistent versus switching frequency

Here are 3 results when ordering 40amp current for different switching frequency values 
Note that the value of the motor current is wrong in the 3 cases ...

m_dc_f_sw = 20kHz
[image]

m_dc_f_sw = 10kHz
[image]

m_dc_f_sw = 5kHz
[image]

abrac, 2019-03-30:
Hi Hug0, can you just elaborate the issue for me? If I understand you correctly, you are expecting the current to be 40A, instead of the 39,37A displayed? I think the small steady-state error can be expected, do you get less error on the output current when connecting a BLDC motor? Maybe it's not a firmware fault at all...

Hug0, 2019-04-01:
Hello, thank you for the interest.
It was over a year ago so I will try to be the most accurate from my memories : 
Here is the context:
Test if the hardware supports constant high current specifications. The power source is a regulated lab power supply 60V-40Amp.
The command is in current and set at 40Amp in motor.
The idea was to check the behavior when DC Switch Frequency is modified (default was 20kHz).

At m_dc_f_sw = 20kHz values are ok (consistent with the drained power from the laboratory power supply)

But when m_dc_f_sw is decreased, the true output power decrease, 420W-> 217W -> 109W as IBatt. Which is consistent with the drained power. 
Or, power don't have to be influenced by the hashing frequency.

I haven't investigated more into the code, but it's as if the value of dc_f_sw, isn't taken into account, or hardcoded in the calculation of I Motor.

I hope I have been explicit enough, 

Hugo

TechAUmNu, 2021-07-23:
This is because at you are turning on the mosfet, turning it off, taking a sample of the current, wait, repeat.
In the time between turning the mosfet on the current will decrease. The rate it decreases depends on the inductance of the  load. 
For a low inductance you would expect the current to fall very quickly, so the if you measure the current only just after you turned off the mosfet and average it, you will see only the maximum current. This is good from a protection point of view as you are effectively setting the maximum current, but it doesn't give accurate average current unless the inductance is sufficiently large.

When you reduce the switching frequency the time between pulses increases, 20kHz -> 10kHz means the current falls twice as much and you will see much lower average current, the vesc will still see the same current.  Same happens at 5kHz and is why you see almost exactly 50% drop in each step you made.

If you tried this test again with a big inductor as a load you would see a very different result. 
This is also why you need to have a suitable inductor on DCDC power supplies and why for low inductance motors you need to increase switching frequency.