vedderb/bldc #59: modular ESC

mkeyno, 2018-04-29:
Hi, sorry to open ticket for my question , actually I intend to made modular ESC in way that controller + gate driver is separated from Mosfets pcb , hence I could increase module ampere by adding parallel Mosfet in the Mosfets pcb, I really appreciated to have your advise or if you know open HW/SW sources let me know

SkyLabRC, 2018-04-29:
Gate driver choice, series resistance and routing between gate driver and fet(s) is application specific and critical to the operational performance. Seperating them to different boards is very tricky:

https://www.infineon.com/dgdl/Infineon-ApplicationNote_GateDriver_MOSFET_PCB_layout_guidelines_for_MOSFET_gatedriver-AN-v01_00-EN.pdf

mkeyno, 2018-04-29:
@SkyLabRC  thanks for info , but honestly I'm not electrical engineering to understand whole technical criteria, however I found following link which use same approach to separate control board and driver board 
http://e0designs.com/products/brushless-rage/
I really interested to have your opinion about that project

closed 2019-02-19 by vedderb