vedderb/bldc #67: Disabling optimizations bricks the device when in BLDC mode

jorticus, 2018-10-20:
So I'm not expecting a solution for this, it's more a warning to others who might try the same thing as I did:

**If you disable optimizations (`-O0`) in the makefile so you can debug via GDB/OpenOCD, and have the firmware set to BLDC mode (or configure it for BLDC mode via the GUI), then the device will hang during intialization and no longer communicate with VESC Tool, effectively bricking it.** 

The only way to recover the firmware is by re-flashing firmware via STLink (which you'd presumably have if you were trying to debug anyway).

Eg. I changed the following in the `Makefile` to `-O0` :
```
# Compiler options here.
ifeq ($(USE_OPT),)
  USE_OPT = -O2 -ggdb -fomit-frame-pointer -falign-functions=16 -std=gnu99 -D_GNU_SOURCE
```
And then uploaded the firmware via OpenOCD and attached GDB to debug it.

I found that the firmware was hanging at this line:
```
mcpwm_init(...) {
  ...
  utils_sys_unlock_cnt();
```

Which I determined was because this suddenly allows ISRs to execute, and `mcpwm_adc_inj_int_handler()` was running for too long too often and starving the main system task, preventing it from executing anything else.

Unfortunately this also blocks communications with the host, resulting in a brick because you cannot configure the device.

If you did want to fix this, I'd suggest finding a way to automatically detect that ISRs are starving the main thread and disable them, then put the device into a fallback mode that just has communications enabled.

The easier solution of course is to not disable optimizations 🙂

vedderb, 2019-02-19:
I'm aware of that, but it is not easy to fix. I think that you definitely should have access to an SWD programmer if you want to do development.

closed 2019-02-19 by vedderb

hackrid, 2019-07-02:
Another option might be to enable optimization only for critical parts using gcc pragmas

https://odellconnie.blogspot.com/2012/02/controlling-gcc-optimization-with.html?m=1

afzalmam, 2020-03-09:
i faced am similar but not exact problem with new toolchain.

On 4.12 hardware, after building firmware using arm-none-eabi-8-2019-q3 & flashing, vesc tool was not able to communicate w/ device, usb uart was not functional.

Connecting the jtag revealed that it was stuck at utils_sys_unlock_cnt(). There is issue #67 (Disabling optimizations bricks the device when in BLDC mode) that mentions hang at the same function, but in that case disabling optimization was causing the issue. But in my case, even w/o 
disabling optimization it happened. Did a hack commenting out to see where it goes, it was getting stuck later at a 1s sleep (iirc) statement.

But upon using gcc-arm-none-eabi-7-2018-q2 there was no issue. In Ubuntu, the version is gcc-arm-none-eabi-7-2018-q3.

My observation is,
following works,
gcc-arm-none-eabi-7-2018-q2
gcc-arm-none-eabi-7-2018-q3

while, gcc-arm-none-eabi-8-2019-q3 doesn't work.

The feeling observation is giving is that w/ GCC 7 it works, but not with GCC 8.

BTW, ARM developer website [1] doesn't contain version used in Ubuntu. Do not know if Ubuntu is going to update gcc-arm-embedded to a newer version, if it does, based on inferences above, there is a chance of issues even with default Ubuntu setup as described in "Setting up a development environment on Ubuntu 16.04 and later LTS releases".

[1] https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads