vedderb/bldc #68: [feature request] ABI encoder support in DC mode

jchillerup, 2019-01-10:
Currently it's not possible to get RPM data in regular non-brushless mode. I'll try to have a go at it myself, so hopefully I'll be able to close my own ticket ;)

vedderb, 2019-02-19:
This should be working in the latest firmware

closed 2019-02-19 by vedderb

abrac, 2019-03-30:
Hi @jchillerup, @vedderb. I am busy trying to familiarize myself with the code - particularly the parts on DC motor. Could you direct me to the file where I can find the change you made to add the ABI encoder support?

I am using a VEsc 4.12 extensively for my university project this year - so you can expect some support from me. :-)

jchillerup, 2019-04-01:
> Hi @jchillerup, @vedderb. I am busy trying to familiarize myself with the code - particularly the parts on DC motor. Could you direct me to the file where I can find the change you made to add the [...]

Check the additions to encoder.c in [this commit](https://github.com/vedderb/bldc/commit/123bb00ab4d8aa2fef8bc9e3c0edb283a1c552ad)