vedderb/bldc #70: Why does the board brake when controller loses connection? Why not coast?

OldNewbold, 2019-02-09:
Why does the ESC brake when losing remote reception? That single feature has caused perhaps hundreds of scuffs, face plants and broken bones. Why not coast to a stop? 

I understand PPM is taken from remote control cars. It is understandable they'd need to brake when signal is lost. But on a skateboard at 35km/h or 20mph, a sudden brake == potential maiming for life...

DieBieEngineering, 2019-02-09:
It doesn't, check your failsafe on your remote / configuration in the VESC.

The behaviour you describe isn't normal.

vedderb, 2019-02-19:
I agree, this is most likely related to the remote. If the receiver issues a full brake command, there is no way for the VESC to know if it did so because it lost connection, or because full brake is used. This is the reason why I never use PPM remotes for longboards - they are simply unpredictable and there is no way to know what is going on from just the pulses it sends. I'm only using my own custom remotes, and hopefully more of them will become available soon.

closed 2019-02-19 by vedderb