vedderb/bldc #75: Signal injection

nitrousnrg, 2019-02-27:
Injecting signals into the control loop will help developing new algorithms and improve the coverage and test automation of the code on every commit.

* Being able to simulate a motor within vesc firmware allows developers and researchers to start from a known motor model and evaluate new algorithms by injecting current and voltages into the control loop.
This is an example of an IPM motor model that could be used to approach issue #26
http://cpp.sh/33vcz

* Light VESC users could easily simulate a motor by launching injection from the command line to learn to operate VESC_tool

* Most importantly, signal injection makes it possible to perform Processor-in-the-Loop testing, executing the code in the actual target and exercising all critical functions for a broader and automated code coverage test, including fault cases as well as normal operating conditions.

rpasichnyk, 2019-02-27:
Being able to simulate a motor would be a dream 😍😍😍

Peemouse, 2019-02-27:
Definitely a great idea !!!

nitrousnrg, 2019-02-27:
We have some basic signal injection working based on that reference code. This issue is a placeholder to reference a future merge taking care of this.