vedderb/bldc #83: User configurable current sensor gain

nitrousnrg, 2019-04-08:
This is more of a question than a request.

I'm aware that commercial VESC board have a fixed current sensor gain because the sensor is integrated on-board, but here at powerdesigns our current sensors are external, attached to busbars and there are many options to choose from, sometimes you are forced to choose different sensors due to stock issues.

Now we have 2 paths to move forward:

1. Expose current sense gain as an user-configurable field within the Advanced tab. Downside is that in most controllers it presents the user a field that is not meant to be modified.
2. Under /build_all/ provide powerdesigns firmware with all our supported sensors. This means each commit with new firmwares will be several MB worth of binaries, which I hope can be avoided.

Or maybe use option 1, and configure tight limits on all the hardware versions, something like:

`#define HW_LIM_CURRENT_SENSE_GAIN			19.98, 20.02
`

That way user's will only be able to tweak current sense gain +/- 1%

Should I go ahead and try making this change?
Thanks!

nitrousnrg, 2019-04-18:
Hi Benjamin, bump about this one. We need a way to support different current sensor gains, I just would need to know how should I move forward, if with many binaries or with user config.
Thanks

vedderb, 2019-04-19:
I thought about it a bit, and one simple way would be to add a terminal command to configure the gain and store it in a hw-specific place in emulated EEPROM. I just added the following functions to the code:

```
/**
 * Read hw-specific variable from emulated EEPROM.
 *
 * @param v
 * The variable to read the result from.
 *
 * @param address
 * Mapped address in EEPROM. Range 0 to 63.
 *
 * @return
 * true for success, false if variable was not found.
 */
bool conf_general_read_eeprom_var_hw(eeprom_var *v, int address);

/**
 * Store hw-specific variable to emulated EEPROM.
 *
 * @param v
 * The variable to store the result in.
 *
 * @param address
 * Mapped address in EEPROM. Range 0 to 63.
 *
 * @return
 * true for success, false if something went wrong.
 */
bool conf_general_store_eeprom_var_hw(eeprom_var *v, int address);
```

I haven't tested the code, so there might be some errors.

The idea would be that you try to load the variable at boot, and if it fails use a default value. Then you can update the variables with terminal commands. The advantage of this is that you can add custom features to hardware without impacting the rest of the code base. The disadvantage is that the terminal interface might be less intuitive.

Can you give it a try and see if it works?

nitrousnrg, 2019-04-20:
Thanks Benjamin, I get it and its a promising feature to implement many other things. I can see a user forgetting to send the command after upgrading the firmware so I hope in the future we can expose the current gain to vesc tool. Perhaps a way to hide settings from vesc tool. Anyway, wishlist stuff.

I will give it a test ride right after finishing the PR I'm working on.

vedderb, 2019-04-20:
Persistent settings between updates would be useful if they are a bit hidden. I will look into storing the settings in RAM before the reboot at the firmware update, and attempt to load them back from there if reading the emulated eeprom fails. Then the function can have an argument that tells if the parameter is loaded from ram, and what the firmware version was when it was stored. Then your hardware file can decide if it is still compatible and store it back if it is. I can probably do something similar for the motor and app settings, and decide if they are compatible based on the signature.

nitrousnrg, 2019-04-20:
Persistent settings would be great for all users, saves some clicks on the GUI :)

nitrousnrg, 2019-04-25:
I tested it today and the store/read functions work well. I have a terminal command that sets the current sensor gain, thanks! This commit is sitting on top of a branch not yet merged, it has to wait for a couple of pull requests.

closed 2025-10-18 by nitrousnrg