vedderb/bldc #98: PPM: Positive Ramping Time does not work

giga993, 2019-08-10:
Hi, 

Hardware is new VESC75/300 R2. 
Software is Vesc-Tool 1.16, just updated the ESC to Firmware 3.58 (at first connection).

Motor Settings are set up by the wizard (FOC), as well as the App Settings (PPM-Remote). 
Everything works like it should.

After that when I switch to to PPM-Tab, (only) change the Positive Ramping Time, Write App Config and Read it again I can see that the changes were written to the ESC (also can read them after power cycle and reconnect, they were definitely written to the ESC). 
Still no matter which Control Mode I use (tried: Current no Reverse, Duty Cycle no Reverse and PID Speed Control no Reverse), when I push the trigger to max, the motor goes instantly to max RPM no matter if Positive Ramping Time was set to 0.1s or 100s (tried some values in between as well, no change in response). 

Am I doing anything wrong, or is this feature not working (anymore)?

Best, 
Chris

giga993, 2019-08-28:
Hi Maxi,
Unfortunately I didn't have time nor access to the hardware the last weeks. I hope I get to it upcoming week.
Somewhere I read that there was a similar problem with interference of the negative ramping time (I'd like to give you the link, but it was some time ago and I can't find it right now, sorry)

My next experiments will include switching the motor direction in software and repeating the tests  (maybe neg. and pos. has some wired behavior when having no reverse and flipped directions) and then trying to changing the negative ramping time at the same amount as the positive to see if I can reconstruct the interference I read about.

vedderb, 2019-09-06:
In current control the motor will always go to max speed without load, even if the current command is small. The ramping time refers to how fast throttle changes will be accepted, so that the throttle can be made less or more sensitive to rapid changes.

closed 2019-09-06 by vedderb

giga993, 2019-09-06:
> In current control the motor will always go to max speed without load
Yeah for sure, because of the low no load current. 
But:
> Still no matter which Control Mode I use (tried: Current no Reverse, Duty Cycle no Reverse and PID Speed Control no Reverse), when I push the trigger to max, the motor goes instantly to max RPM no [...]

No matter which control mode I use, the ramping does not work like it should. After changing the Negative Ramping Time as well, the Ramping works (tried Duty Cycle no Reverse), but takes the Negative Ramping Time for positive and negative ramping. Which is not useful for my case, I need a high positive ramping time for smooth ramp-up and a instant ramp down for safe stop.