vedderb/bldc #100: canbus status message frequency at low speed

Matrixgta, 2019-08-21:
I face a strange behavior with status messages over canbus, selecting all 5 messages and frequency from app config I read over can only few and random status packet even setting high frequency, messages will increase as controller duty cycle increase, even if set point comes from canbus or from app command, at duty >80% i read on the bus all 5 messages but they stop if duty is set to 0.

vedderb, 2019-09-06:
That sounds very strange indeed. Does anyone else have a similar issue?

Matrixgta, 2019-11-18:
After further test comes out that the problem is somehow my hardware (Trampa VESC) , the canbus termination resistor have wrong value and this cause the problems, still did not find how this can be related to duty but with corret resistor value messagess are now at the right frequency

closed 2019-11-18 by Matrixgta