vedderb/bldc #106: BMS functionality?

mikepurvis, 2019-09-24:
I'm wondering how realistic it might be to add some code to VESC for basic fuel gauge estimation. It already accumulates Ah and Wh spent while in operation, so I think the missing pieces would be:

- Persisting the information through power-cycles, in some kind of flash-based circular buffer.
- Configurable logic to reset the state for when the battery is found to be full or empty based on voltage extremes.
- Using one of the ADC inputs to monitor the charge lines. This would require a bit of extra wiring so that the uC would be powered up when the charger was connected, but doing this extra wiring would still be easier and cheaper than integrating a separate gas gauge circuit.

The current functionality is pretty good if you always charge to full and do your own book-keeping as far as remembering how much you've used if you use your system multiple times on a single charge. However, it would be great to have the final pieces in place for this.

Thoughts on feasibility?

Peemouse, 2019-09-24:
Hi @mikepurvis,
The 3 points that you explain are the role of the BMS.
What you'd be interesting is to develop kind of a standard protocol with BMS connected through CANbus (such as [DieBieMS](https://github.com/DieBieEngineering/DieBieMS) ) for a soft cut out in case of battery dead or other faults.
We work on this [here](https://www.electric-skateboard.builders/t/flexibms-lite-new-approach-to-get-past-vaporware-stage/85493/244?u=pimousse) and we'll propose a PR for this.

mikepurvis, 2019-09-24:
Oh yeah, I'm aware I'm proposing functionality normally found in a discrete BMS; hence the title of the ticket. However, I'm pointing out is that VESC already has the most difficult piece of that functionality done, which is monitoring and accumulating the consumed wattage. It's a fairly small step from there to supplying users with a fully functional gas gauge, accessible over the UART/CAN interface, and via peripherals such as the Wand.

Given that VESC now also includes such extras as an IMU for UAV use-cases, it seems reasonable to consider battery tracking for inclusion given the value it would provide to users in avoiding yet another component to have to integrate, and deal with interop issues on, etc.