vedderb/bldc #109: Parameters reset after reboot

nyxrobotics, 2019-10-03:
I am using VESC-MINI4 on Ubuntu16.04 and vesc-tool.
My VESC-MINI4 works well at firmware 3.40.
Today, I updated my VESC-MINI4 to firmware 3.61.I set parameters and write to rom with vesc-tool, but all parameters reset after DC power off.
This problem happens with  firmware 3.60 and 3.61 and 3.62, but won't happen with firmware 3.40.
Do you have a good idea?

Teslafly, 2019-11-15:
I have experienced this too on my dev setup of a stm32 discovery board glued to an off the shelf powerstage. If I figure it out I'll commit a fix. My guess is it may actually  be at the root a hardware problem though

AmirHusseinZia, 2022-05-10:
> I have experienced this too on my dev setup of a stm32 discovery board glued to an off the shelf powerstage. If I figure it out I'll commit a fix. My guess is it may actually be at the root a hardware problem though

did you fix the problem?

Teslafly, 2022-05-10:
> > I have experienced this too on my dev setup of a stm32 discovery board glued to an off the shelf powerstage. If I figure it out I'll commit a fix. My guess is it may actually be at the root a hardware problem though
> 
> did you fix the problem?

This was fixed in the current dev firmware. In the discovery's case, it operates at 3.0v, which is not enough for the fast flash write. You need to scale the flash write speed/size by the mcu voltage. If your mcu operates at 3.3v it is unlikely this is your issue.

AmirHusseinZia, 2022-05-11:
I solved this problem.
First you have to upload the bootloader with ST-link and then flash the firmware using vesc tool (I've been flashing the firmware with ST-link and this was the problem)