vedderb/bldc #118: Custom app configuration

ddosoff, 2019-12-09:
Hello Benjamin,

Thanks to https://github.com/vedderb/bldc/blob/master/applications/app_custom_template.c it was easy to start implementing my custom app.

But there is no easy way to work with my custom app configuration. I'd like to add page to vesc_tool the same easy way as implementing custom app in bldc.

If I modify https://github.com/vedderb/bldc/blob/master/datatypes.h#L606 app_configuration it breaks compatibility with original vesc_tool.

I think simplest way is implement terminal commands to update custom app configuration.

Could you point correct way to save and restore config struct of custom app to EEPROM?
May be you have some ideas how to solve this in more elegant way?

mhweiner, 2019-12-20:
I'm first getting into this and potentially looking to become a contributor and make my own 3rd party app integrations. However, for what it's worth, I think this is a great idea and would love to see this Issue get some attention.

vedderb, 2019-12-21:
There is a way to store custom variables persistently for your app. You can use

`
bool conf_general_store_eeprom_var_custom(eeprom_var *v, int address);
`

and

`bool conf_general_read_eeprom_var_custom(eeprom_var *v, int address);`

Each eeprom var is a 32-bin union that you can use to store float, uint32 or inte32. You can store up to 64 different custom variables. The above functions also take care of slowing down the watchdog and stopping the motor when used, as the CPU will halt while writing to flash.

To start with you could edit the settings with terminal commands, but I'm planning to make it possible to load qml plugins in VESC Tool at runtime to make a custom gui for your application. The GUI could either use terminal commands or these functions to send a custom byte array:

`void commands_set_app_data_handler(void(*func)(unsigned char *data, unsigned int len));
void commands_send_app_data(unsigned char *data, unsigned int len);`

Using this approach does not break compatibility and should be easy to maintain. It might also be possible to compile the QML into the VESC Firmware compressed with LZO to make it appear on any VESC Tool when starting the custom app without loading it manually; maybe that is impractical though,

ddosoff, 2019-12-21:
Thanks, that is exactly what I need.

I found https://github.com/vedderb/vesc_tool/blob/master/application/create_app to create my custom app.

This is very convenient for me too. Will create different app, because interface is too different.

Good idea to use common connection page in the future. I will create PR when application becomes ready.

closed 2019-12-21 by ddosoff

mhweiner, 2019-12-21:
@vedderb Awesome, thanks! By the way I'm a huge fan :) Thanks so much for all your inspiring work!