vedderb/bldc #119: 3.64 can_scan breaks with ppm enabled.

fessyfoo, 2019-12-22:
[ originally posted here: https://vesc-project.com/node/1501 not sure right place to go with this type of issue] 

HI, I upgraded to vesc 3.64/vesc-tool 1.27 on a dual foxbox (legacy) setup connected via can bus.

my default secondary can id is 84. primaray is 21.

can_scan identifies 84. however if I enable the PPM app or the PPM + UART app then the can_scan will show the secondary device as id 93 or sometimes 92. can forwarding fails. on all 3 ids, 84, 93, and 92. changing the remote can id to 1 and master to 0 has similar effect. with the secondary being seen as id 9 or 10 instead of 1.

downgraded to vesc 1.62 / vesc-tool 1.25 and cannot reproduce the problem.

```
// PPM enabled
Found VESC with ID: 92
Done

// just rerun can_scan
Found VESC with ID: 93
Done

// PPM app disabled.
Found VESC with ID: 84
Done
```

vedderb, 2019-12-22:
That sounds very strange. I just tried it myself with two VESC6 connected over USB and UART and can't reproduce the issue. Can you try loading the default motor and app configuration on both focboxes with the new firmware, write the configurations back and then only switch on ppm?

Visnu7777, 2019-12-22:
Anyone else having issues with latest app and firmware? I just did an update, on the two first attempts I got my first firmware update timeouts ever. After power cycling I tried to update all 4 separate, also over can over the master that has a metr pro. After that worked I could successfully do the motor setup for all 4 drives, but soon after (still on the bench) one drive did not respond any more. When I tried to scan can bus only one vesc shows up but when I try to connect it can’t. I’ll report back when I opened the case to use USB to check what’s wrong… Now only the two drives in the back (master and the other on the pcb) respond to the remote (which means can is working somehow) but I can’t connect to that one. (copied over from my post on esk8.news) reverting to 2.63 solved it.

vedderb, 2019-12-22:
I think I found the issue. When updating app_ppm I used ST2MS instead if MS2ST in the sleep function, which probably prevented it from sleeping and thus starving other threads. This is fixed in FW 3.65 and VESC Tool 1.28, which should finish compiling in about 20 minutes. Can you give it a try?

Visnu7777, 2019-12-22:
Now can scan and connect works but motor detection fails with a generic timeout and one time can timeout. Detection looks good though, one unit doesn't blink red during the detection. (4wd, 2x ebmaker dual mini v. 4.10 hw)

fessyfoo, 2019-12-23:
> Can you try loading the default motor and app configuration on both focboxes with the new firmware, write the configurations back and then only switch on ppm?

Yeah, this is all i did to reproduce,  used default config, only settings change was enable/disable PPM. 

Trying the new 1.28/3.65 next.

fessyfoo, 2019-12-23:
Ok, 👍 the weird can_scan behavior appears to be fixed in 1.28/3.65.

Thanks.

closed 2019-12-23 by vedderb