vedderb/bldc #121: USB/Bluetooth disconnect events should be handled to avoid blocking in the commands_printf()

ddosoff, 2019-12-26:
Hello Benjamin,

My app use `commands_printf()` just after start. It prints log like events and I think this is good idea overall.

Right after boot `send_func_blocking` is null and this prints just ignored and it is correct.

Then I connect USB cable. After first terminal command send_func_blocking updates to comm_usb send_packet_raw and I could see my app log. I am happy.

If I disconnect USB cable, my app works a bit more and then locks in the `commands_printf()` when USB driver buffer becomes full. If I connect USB cable again driver sends me buffer, I see last output and my app works again.

I think this is not correct behavior. In case of USB disconnect event `send_func_blocking` should be cleared to ZERO again not to block `commands_printf()`.

I did not found any code to reset `send_func_blocking` pointer.

Please tell me your opinion if it should be fixed and recommended way to fix it.

I'd like to fix it myself, but I have no SWD programmer and afraid to break my USB connection to VESC.

Please help.

ddosoff, 2019-12-28:
Another words it is not clear that `commands_printf` could be blocked forever if connection disconnected.

vedderb, 2020-01-30:
I noticed that too. USB now has a timeout and throws away the data, so commands_printf should no longer block.

closed 2020-01-30 by vedderb

ddosoff, 2020-01-31:
Now fixed. Thanks!