vedderb/bldc #122: DRV8323/s/8353 SPI Bug

Jfriesen222, 2019-12-31:
The DRV8323 SPI interface is not live when the enable pin is low. The current FW tries to write settings for current limits etc. before enable pin is driven high, so the settings are never written. 

 Additionally I notice a small bug with inconsistent SPI writes unless I add a short delay after the CS pin is driven high (one additional spi_delay) before the spi transfer.

vedderb, 2020-01-30:
Thanks, I have added the delay. I don't think the SPI is used when enable is low.

closed 2020-01-30 by vedderb

Jfriesen222, 2020-01-30:
drv8323s_init() is called before the enable pin is high and it attempts to write a control register. Additionally  in mc_interface the ifdef for drv8323 which contains spi writes to control registers occurs before the enable pin is driven high in mcpwm_foc_init. A workaround is to drive enable pin high in the hw_init function but this means you drive it high before initalizing timers which seems undesirable.