vedderb/bldc #126: Universal motor temp sensor

ddosoff, 2020-01-21:
I created polynom for KTY83/122 (click on the page in the bottom):

https://docs.google.com/spreadsheets/d/1iJA66biczfaXRNClSsrVF9RJuSAKoDG-bnRZFMOcuwU/edit?usp=sharing (huge error on PTC linear formula)

And set coefficients for only this temp sensor here: https://github.com/vedderb/bldc/blob/master/mc_interface.c#L1785

But idea that **this 4 coefficients could be set from vesc_tool**.

It is possible to support any temp sensor without firmware changes.

Please share your opinion.

vedderb, 2020-01-30:
I think that would be nice, but I want to do it in combination with a page in VESC tool where the coefficients are calculated and plotted, to avoid getting it wrong when messing with spreadsheets. We can look into this some time.

ddosoff, 2020-01-30:
It is possible to implement polynom calculation into UI, but time consuming. This will not help in sharing calculated coefficients for different temp sensors.
The important thing is sharing this curves. I think just good instruction how to calculate it and share is better overall.

Now it is not easy to add enum values for temp sensors or battery chemistry. I'd like to know battery chemistry to calculate battery percentage known cell voltage. I have LiNMC batteries and I have to modify battery type enum.

New enum value require 2 pull request into bldc and vesc_tool and increment in version. I do not know good solution yet, but will think how to simplify it.

vedderb, 2020-01-30:
You can also just make a pull request to the firmware, and I will fix the VESC Tool stuff. It is quite easy for me to accept the request, merge it with my local changes and add the enum to VESC Tool.

ddosoff, 2020-01-31:
Yep, but my PR with enum for KTY83/122 https://github.com/vedderb/vesc_tool/pull/59 still not merged.

I have to divide futures into different branches. Please left a comment what I have to fix in PR to apply it.

Because I fixing this PR with each new vesc_tool version.

vedderb, 2020-01-31:
The enum is in the latest config in VESC Tool, I added it manually, so it should work. I'm not sure about the application structure change in the PR though, I have to think about it more. Copying the files makes the application independent, and does not require putting the files relative to the VESC Tool path.

ddosoff, 2020-01-31:
I am sure, that copying is worse then relative paths.

The advantage of copying that application is always buildable, but merging with latest vesc_tool becomes a horror.

With relative paths just some minor merges could be required. And usually not changes at all.

vedderb, 2020-01-31:
This is probably the wrong place to discuss this, but I don't want to figure out how to move things around right now :-)

I don't agree here. The idea is that the VESC interface has the features you need, and if you miss something it is best to add add them to VESC Tool first. If you want to do something special that needs to be done in the interface and has no other general use worth including in VESC Tool, it is probably best to subclass the classes rather than making a special version that needs manual merging for every update of VESC Tool.

First I thought of building VESC Tool as a library and linking to it, but then the library needs to be compiled on every platform that your custom application is going to run on. I like the fact that I can move everything to the raspberry pi and compile it there without having a set of external dependencies or dragging around a bunch of files I'm only using a fraction of, that have to be in a specific relative path to my application.

To sync with the latest version I would like to have some script that replaces the VESC files without changing the application file. Maybe it would make sense to put all application files in a separate pri file as you did to avoid having to change the pro file, but I don't want to have a pri file and a bunch of other files that have to be in a separate path away from my application.

ddosoff, 2020-02-03:
I agree, that binary library is not necessary. But why you do not like separate pri file still is not clear?

I think both ways possible. For my app I'd like to use this pri file with relative paths.

Pri file is not break anything in the vesc tool. Why not?