vedderb/bldc #131: VESC keeps resetting when built using  gcc-arm-none-eabi-9-2019-q4-major

gowrav, 2020-02-06:
Found the issue initial with MAC and Raspberry Pi but rechecking the gcc-arm-none-eabi version shows the build is stable and functioning when using gcc-arm-none-eabi-7-2017-q4-major but keeps resetting with flashing RED and GREEN light every 1 second when the firmware is built using the gcc-arm-none-eabi-9-2019-q4-major

Hitch:
The issue exists when the conf_general.h file is modified i.e when the hardware is changed from 
//#define HW_SOURCE "hw_60.c"
//#define HW_HEADER "hw_60.h"

to 

//#define HW_SOURCE "hw_75_300.c"
//#define HW_HEADER "hw_75_300.h"
or
#define HW_SOURCE "hw_a200s_v2.c"
#define HW_HEADER "hw_a200s_v2.h"

The default untouched source builds completely fine., but is unusable as the hardware is different.

romixlab, 2020-03-07:
+1 same thing, but with 2019 version. Interestingly with a little bit older 2019 version everything seemed to work fine for several days before flashes. With the newest version it crashes immidiately. Using 2016 right now.

afzalmam, 2020-03-09:
i faced am similar but not exact problem with new toolchain.

On 4.12 hardware, after building firmware using arm-none-eabi-8-2019-q3 & flashing, vesc tool was not able to communicate w/ device, usb uart was not functional.

Connecting the jtag revealed that it was stuck at utils_sys_unlock_cnt(). There is issue #67 (Disabling optimizations bricks the device when in BLDC mode) that mentions hang at the same function, but in that case disabling optimization was causing the issue. But in my case, even w/o
disabling optimization it happened. Did a hack commenting out to see where it goes, it was getting stuck later at a 1s sleep (iirc) statement.

But upon using gcc-arm-none-eabi-7-2018-q2 there was no issue. In Ubuntu, the version is gcc-arm-none-eabi-7-2018-q3.

My observation is,
following works,
gcc-arm-none-eabi-7-2018-q2
gcc-arm-none-eabi-7-2018-q3

while, gcc-arm-none-eabi-8-2019-q3 doesn't work.

The feeling observation is giving is that w/ GCC 7 it works, but not with GCC 8.

BTW, ARM developer website [1] doesn't contain version used in Ubuntu. Do not know if Ubuntu is going to update gcc-arm-embedded to a newer version, if it does, based on inferences above, there is a chance of issues even with default Ubuntu setup as described in "Setting up a development environment on Ubuntu 16.04 and later LTS releases".

[1] https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads

afzalmam, 2020-03-10:
Realized later that gcc-arm-embedded package is not maintained by Ubuntu, but a PPA (my Linux system is not Ubuntu), so seems the problem will only happen if those maintaining the PPA update to newer toolchains that has issues.

gowrav, 2020-04-05:
Can be resolved by using gcc-arm-none-eabi-8-2019-q3-update or the version specified by README file.

closed 2020-04-05 by gowrav

romixlab, 2020-04-07:
> Can be resolved by using gcc-arm-none-eabi-8-2019-q3-update or the version specified by README file.

Do you know what causes the problem? would be good to know that, because it is easy to switch compilers in this particular case, but this may not always be the case.