vedderb/bldc #140: re-scaling CAN bus setDUTY CYCLE from -1 to 1 instead of 0 to 1

abdowx, 2020-03-09:
Hi everyone,

I just started using Vesc tool few days ago, and I would like to drive the motor in both senses with same duty cycle CAN frame, so for a 0 payload frame I would like a "-1" duty cycle.

I'm not expert in coding, but I tried to re-load a custom firmware after modifing the code line in "comm_can":

buffer_append_int32(buffer, (int32_t)(duty * 100000.0), &send_index);

by:   buffer_append_int32(buffer, (int32_t)((duty+1) * 50000.0), &send_index); 

After loading the firmware, the modifications of the function implemented do not seem to be taken into account.

If someone could clarify me a bit on the problem, I would be very grateful.

Thank you :)

(I have a hardware version 60 and I did also defined it on conf_general.h "#define HW_VERSION_60")

Peemouse, 2020-03-09:
Hi,
Did you the same on the other side (reading) ?
You send a modified value but does the can comm know how to handle this "new" value ?

abdowx, 2020-03-11:
First of all, Thank you for your reply,
For me and if I understood you correctly, there should be a line of code on another source file which the formula for CAN-RX '(CANvalue)=(duty*100 000)' to handle my value sent in CAN of 10,000 for example.
But after looking after it in all the code files where we call "duty" I can't find it out.

Peemouse, 2020-03-14:
You need to apply your changes here as well : 
https://github.com/vedderb/bldc/blob/e1c859c894bf88b082e589234dc8c78125b9fb8e/comm_can.c#L837