vedderb/bldc #157: Possible miss behavior in correct_hall()

ddosoff, 2020-04-18:
Hello Benjamin,

I checked the code with PVS-Studio static  analyzer.

Here are some points:

https://github.com/vedderb/bldc/blob/master/mcpwm_foc.c#L3344 - Expression 'ang_hall_int_prev == - 2' is always false.

Because here https://github.com/vedderb/bldc/blob/master/mcpwm_foc.c#L3333 type of m_conf->foc_hall_table is **unsigned**.

vedderb, 2020-04-18:
ang_hall_int_prev is assigned to -1 and -2 just below when not using the hall table, so that code is correct. The hall interpolation is updated  a lot it FW5 anyway, so this code is outdated.

closed 2020-04-18 by vedderb

nitrousnrg, 2020-04-18:
The static analyzer was right, I found that line a few weeks ago (https://github.com/vedderb/bldc/pull/143#discussion_r395994430)

Keep the discoveries coming!

vedderb, 2020-04-18:
Looked at the old code now, and the assignment before the if statement should have been after it, which is the actual mistake. Anyway, this function is completely updated in fw5, so it does not matter.

ddosoff, 2020-04-26:
I tested fw 5.0, seems really good improvement in hall positioning. Thanks!