vedderb/bldc #169: Incorrect temperature in ESC status in UAVCAN mode

jxltom, 2020-05-15:
In https://github.com/vedderb/bldc/blob/b460fa0fa3cb507a14aef26082d782f76b30d7db/libcanard/canard_driver.c#L119, the returned temperature is always ```298.0000``` which looks like lost precesion.

Even I changed code to ```status.temperature = 24.9219 + 273.15;```, the parsed temperature in https://github.com/UAVCAN/gui_tool is still ```298.0000```

Maybe something wrong with ```canardConvertNativeFloatToFloat16```?

@pavel-kirienko do you mind also have a look? Thanks.

jxltom, 2020-05-15:
Change to float32 is fine. It looks like using float16 is overflow or something like that for big numbers such as ```280```.

pavel-kirienko, 2020-05-19:
At this exponent, the step is around 0.2:

```python
>>> import numpy
>>> numpy.nextafter(numpy.float16(298), numpy.float16(1000))
298.2
```

Whereas your value falls closer to 298:

```python
>>> 24.9219 + 273.15
298.07189999999997
```

The behavior you are observing is correct.

jxltom, 2020-05-20:
Thanks for the clarification!

So the precesion of ```float16``` for numbers such has ```298``` is around ```0.2``` or bigger.

Since uavcan is using Kelvin and motor temperature is certainly larger than ```273.15```, maybe it makes sense to change the definition of ```uavcan.equipment.esc.Status.temperature``` to ```float32```, otherwise the digit ```0.05``` in ```273.15``` will be definitely lost. ```float32``` doesn't have this issue.

```
>>> import numpy
>>> numpy.nextafter(numpy.float16(273.15), numpy.float16(numpy.inf))
273.5
>>> numpy.nextafter(numpy.float32(273.15), numpy.float32(numpy.inf))
273.15002
```