vedderb/bldc #172: Dynamic switching frequency (FOC)

javmarina, 2020-05-17:
I've seen in the source code that in BLDC mode the switching frequency is dynamically changed according to speed. However, in FOC, when motor configuration changes and those changes include the switching frequency (foc_sw), the timers are reset. What could happen if we directly modified the ARR register in the timer? I want to change the switching frequency based on current speed but I'm not sure how it can be achieved.

Thanks in advance.

TechAUmNu, 2020-06-14:
I found that changing the switching speed on FOC can significantly change the detected motor parameters. So the parameters would need to be measured over the full range of switching and accounted for.

nitrousnrg, 2020-06-14:
> I found that changing the switching speed on FOC can significantly change the detected motor parameters. So the parameters would need to be measured over the full range of switching and accounted for.

This would deserve opening an issue on its own. Parameter detection results vary wildly with different switching frequencies and different powerstage turn on/off times. 

About dynamic frequency... I guess its doable. Maybe the current controller PID expects a fixed frequency, idk. I see the benefit on lower switching losses so let us know if you get to try it.