vedderb/bldc #173: Batterylevel not right under high load

gianmarcov, 2020-05-19:
Hi,
I controll an ebike hub motor with 10s5p max. 40A of current and it works nice in FOC mode, i show the battery level on the propertairy display.

At my first version i got the battery voltage from the vesc and show it on the display, but under load it flips up and down, this was not so helpfull, so i changed it to the function mc_interface_get_battery_level but it did the same, the only change was it return 0-100 as float. So under load i cannot use the battery voltage for getting the battery level.

https://github.com/vedderb/bldc/blob/9781fb95636e9b18a51d584ae62928493847d370/mc_interface.c#L1347-L1355

So i created, this little algorithm / logic to calculate the voltage when the motor is under load and it seems to be better than use the input voltage, i tested it over 12 Km and 30Km/h AVG and it works like desired, so when the motor stops to assist the battery level does not change, so the accuracy is ok.

when the motor is under load:
```
        Vesc::getAmpHours(&ampHours);
        batteryLevel = (vampHours - (ampHours - offsetAmpHours)) / BATTERY_AMPS_HOUR_FACTOR;
```
when the motor isn't under load:
```
        Vesc::getAmpHours(&offsetAmpHours);
        Vesc::getInputVoltage(&openCircuitInputVoltage);
        openCircuitInputVoltage = openCircuitInputVoltage < MIN_BATTERY_VOLTAGE ? 
        MIN_BATTERY_VOLTAGE : openCircuitInputVoltage; // v_min = 32V
        batteryLevel = 10 * (openCircuitInputVoltage - MIN_BATTERY_VOLTAGE);
        vampHours = batteryLevel * BATTERY_AMPS_HOUR_FACTOR; //  factor = 14.5 / 100 = 0.145
```
I would create a pull request with this version if it's ok.
I can create another option 'BATTERY_TYPE_LIION_3_0__4_2_AMP_CALC' or i can change the 'BATTERY_TYPE_LIION_3_0__4_2' code?

Another question is to add a command to reset the ampHours or when the vesc shutsdown that it resets all logging values.

Mitchlol, 2020-06-11:
Maybe you can use the motor cutoff voltage to help rather than 3.2

gianmarcov, 2020-06-22:
@Mitchlol 
cutoff has nothing to do with the issue.

When reading the battery level i got the level calculated by the actual voltage.

The calculated battery level is right only when the battery has no high load attached to it. When the load is attached, the battery voltage drops 1-4V down showing the wrong battery level.

My fix is to calculate the consumed Ah with the given formula and get the battery level predicted. Then when no load is attached, the battery level is calculated normaly via the voltage.

So the issue is:
no load = 42.00V = 100%
load 10s = 40.00V = 80%
after load = 41.80V = 98-99%
load 10s = 39.00V = 70%
after load = 41.60V = 97-98%

When i drive 10 min. under load i well see the wrong battery level util i stop.