vedderb/bldc #181: Make VESC Protable to "any" Chibios supported MCU

Tecnologic, 2020-06-09:
Hi Guys,

i have here a hardware equipped with the STM32F446RE, which i thought could be nearly compatible with the used F405/7 but thats was not the case...

So i updated Chibios from version 3 to the latest release v20. See [https://github.com/UNIMOC-Universal-Motor-Control/vesc-software/tree/chibios_portability](https://github.com/UNIMOC-Universal-Motor-Control/vesc-software/tree/chibios_portability) for reference.

I now need to set the memory layout in the linker file. VESC original Linker File uses the following Flash layout. 
`    flash : org = 0x08000000, len = 16k
    flash2 : org = 0x0800C000, len = 393216 - 16	/* NEW_APP_MAX_SIZE - CRC_INFO */
    crcinfo : org = 0x0805FFF0, len = 8		/* CRC info */`

Could some one explain the sections to me. For what are the first 16k used? Bootloader? 
Why does the flash2 not utillize all of the remaining flash? Is this because of the configuration storage? What else is is important to know about the flash layout?

Regards

Tec

Tecnologic, 2020-06-11:
Firmware runs on F446 but when I try to connect with vesc tool 2.06 I get the error that it can not read the firmware version. In the firmware I can see that it triggers the right command in the comm and sends back I packet with firmware version 5.02. 

Any ideas appreciated.

Thanks 

Tec

nitrousnrg, 2020-06-12:
no clue what could be going on, but as a random thought... you may want to stick to a GCC version that we are sure works for vesc. AFAIK all the testing is done in gcc v7, and there were some hard to chase issues with particular versions of gcc v8.
https://github.com/vedderb/bldc/issues/84#issuecomment-490600780

Tecnologic, 2020-06-12:
I will trace the traffic on the serial and have a look what's transferred.

Tecnologic, 2020-06-16:
[image]
I needed to update the usb handling as well. the usb config for chibios was missing several stuff which was added in the meantime. Now it Works!!! 

Now i need to configure the hardware properly. I will keep u updated.

Prevas-ERRA, 2020-06-17:
Very good, but for portability to F7, H7 and other you also need to change drivers.
From ST SPL that are not available for new families to ST LL (low level drivers, part of CubeMX).
https://www.st.com/en/embedded-software/stm32-standard-peripheral-libraries.html
And/or use more ChibiOS drivers to be able to use other than STM32 chips.

Tecnologic, 2020-06-17:
I know. But when i do all that stuff at once the pull request will be to huge to handle. My goal now is to have my hardware running and get all the other hardwares to build so that others can test this. At this point i would open a pull request. 

Next Step would then be to get rid off the SPL and only use CHibios HAL. From what i know now of the code this is perfectly possible because all used SPL drivers are available from Chibios aswell.  But for all of this u need to change the hw configurations layout.

CTSchorsch, 2020-06-19:
I like the separation of hardware config files and seperate makefiles. i use similar constructions for cross platform projects

do you plan a "master" makefile to do thinks like "make hw60" ?

Tecnologic, 2020-06-19:
I have not planned that but it's a good idea.

Juanduino, 2020-09-10:
Hi.

First of. Great work on this, All of this. You people rock! EV future here we come!

Will this port work for the Teensy 4.0? How will a go about making it work. Any pointers would be greatly appreciated. 

ChibiOs is running on T4: https://github.com/greiman/ChRt

Juanduino, 2020-09-10:
I got the answer I was looking for. No need for T4, STM32 works just fine 👍 

https://github.com/greiman/ChRt/issues/12

Tecnologic, 2020-09-10:
Hi,

basically it can work BUT there would be a lot of work needed to rework all the ST standard peripherial library references in the code. I stopped that effort because the support for STM32F446 is fine for me.

TechAUmNu, 2020-09-22:
Thanks a LOT for doing this, I am about to start porting the code over to run on the STM32H743, Its an M7 running at 480MHz. 
If I only need to rework the use of the peripheral libraries then that will save me a lot of time :D

Tecnologic, 2020-09-22:
Hi Tech, yes the old st lib is the only dependency which prevents that. Please keep me updated.

TechAUmNu, 2020-09-22:
Just been trying to get this setup in an IDE, looks like you used STM32CubeIDE? 
Do you have a project file for it? (Not sure what folder I should be pointing it at to get it working properly)
I see the make files for each hw so I guess you added each hw folder as a project?

Sorry for a million questions, I am used to Keil so eclipse is a bit jarring!

Tecnologic, 2020-09-22:
The ide should not matter. I used stmcubeide with the project files in the top level folder of the project. Then I used build configurations with change the base folder for make to the current hardware configuration folder. To build eclipse just called make -j all in that folder. 

I can supply u withe the project files next week.

TechAUmNu, 2020-09-22:
No worries, got it all working now and can debug on my nucleo board. Will keep you updated on my progress.

Tecnologic, 2020-09-22:
Great. Thanks for trying it.

magicrub, 2021-05-23:
ArduPilot would love to interface with this project.

Teslafly, 2021-05-24:
> ArduPilot would love to interface with this project.

Vesc already supports uavcan if ardupilot does.

This issue is more about running the motor controller code on mcu's other than the stm32f405/7

Tecnologic, 2021-05-24:
F446 works for others more work is needed. All the references to spl should exchanged with chbios Hal drivers.

magicrub, 2021-05-24:
> > ArduPilot would love to interface with this project.
> 
> Vesc already supports uavcan if ardupilot does.
> 

We want to run VESC as a library inside the autopilot

BrokuLP, 2024-02-11:
Could anyone update me on the status of this issue. I might be interrested to continue the port to a new chibios version.

Tecnologic, 2024-02-11:
The problem for a complete port is the usage of old st spl drivers. E.g. the hole flash eeprom stuff would need to be ported to mfs in chinos. With all this U have the risk to break the firmware for a lot of special Hardwares. I was mainly lacking time and test hardware.

TechAUmNu, 2024-02-11:
What is the latest version of your port? I understand the core of vesc a lot better now so want to take another stab at it. Especially now that we have LISP and there have been several times when I really needed faster control for high erpm motors. Having the ability to go up to a much higher performance chip would be great.

Tecnologic, 2024-02-11:
I think a v22 of chibios. Just give it a shot with just updating the chibios folder. The compiler will point U to the obvious stuff.

TechAUmNu, 2024-02-11:
Updated it to 22.11.3, just needed the config version updated.

TechAUmNu, 2024-02-12:
Merged in latest vesc master and removed the ST SPL. Now the fun begins to fix a bazillion errors.
https://github.com/TechAUmNu/bldc/tree/chibios_update

Tecnologic, 2024-02-12:
Start with the makefiles and the chconf.h, mcuconf.h halconf.h these need an update from the ones in the f4 demo. That brings U down in errors by a lot

BrokuLP, 2024-02-12:
> Merged in latest vesc master and removed the ST SPL. Now the fun begins to fix a bazillion errors. https://github.com/TechAUmNu/bldc/tree/chibios_update

Ok, when I have Time I will take a look at it

TechAUmNu, 2024-02-13:
For now I just enabled the SPL again to get it working properly before changing drivers. 

It now compiles, with a few warnings from the encoder stuff. 
I put it onto my board and the leds are flashing, usb shows up, but vesc tool doesn't receive anything on the port.

Tecnologic, 2024-02-13:
For vesc tool I changed something. Have a look on the commits from my branch.

Tecnologic, 2024-02-13:
Reopen because of new interest.

TechAUmNu, 2024-02-13:
> For vesc tool I changed something. Have a look on the commits from my branch.

I don't see anything there that should affect it (assuming you mean this https://github.com/UNIMOC-Universal-Motor-Control/UNIMOC_GUI)

Tecnologic, 2024-02-13:
No I mean in the vesc controller code. I know there was something but I can't remember what. Will have a look tomorrow.

TechAUmNu, 2024-02-13:
Found it, the increment that says the usb has been configured got deleted in the merge of that file. Now its working fine running motor and everything. 

I think the only bit that is broken going by the warnings is the encoder stuff.

Will branch this now to have a working version before starting to remove the SPL

Tecnologic, 2024-02-13:
Sounds like a good approach. Then one by one each spl driver.

Scarittagle, 2025-10-31:
@Tecnologic @TechAUmNu hey how you guys doing so far on the porting? I'm interested on porting this to the G431 mcu, seems like a nice reference for me