vedderb/bldc #182: Hall sensors need more filtering

blezalex, 2020-06-12:
VESC v4 used to have low pass filter at hall sensor input. V6 and 75/300 don't have those filters anymore.

There is a software filter, averaging 3 samples, which is not sufficient in many cases. 5 out of 3 motors I run, suffer from sync issues. (those motors are of different kind, 18 inch hub motor, 11 inch hub motor and 6354 outrunner, skate hub motor, onewheel +/xr hub motor)

The issues manifest themselves as "clicks" or loud bangs when motor is under load and sharp loss of torque. I believe the EMF from phase wires is picked up by hall sensor wires and causes hal sensor error. This issue is especially critical in balancing application, as loss of torque causes a crash and nose dive.

To confirm that hypothesis, i added a basic check (see code change https://gist.github.com/blezalex/52f4026cfc891c16771fc89749806a1e)

I can see lots of errors which happen under load. Note that this check captures only the most obvious errors hall=0 or 7, and does not catch many others.

Changing filtering from 3 samples to 7 solves the problem for my motor, but probably not for all.
https://gist.github.com/blezalex/952f01b0cb11339916d9cceb4403d064

Ideally we need a hardware solution (switchable hall low pass filter) or configurable low pass filter in software that takes more samples into account. 

I'm running my motors with patched vesc software, but I guess other users would benefit form this fix as well.

blezalex, 2020-06-12:
I made a circuit board with an external Low pass filter, that also solves the problem. If vesc shipped with board like that, that would have been an acceptable solution as well.

TechAUmNu, 2020-06-14:
What RC did you use for your filter board?

blezalex, 2020-06-14:
R=100om 
C=100nF

Initially i wanted R higher but high values would have issues with vesc pullup resistor (2.2k)

nitrousnrg, 2020-06-14:
Watch out for phase shift introduced by the lowpass filter
[image]

you don't want to introduce more than 3° of angle error due to the RC time constant. This dumb AC approximation tells me you shouldn't rely on that setup above 8000 erpm.

Hall inputs are usually ignored at high rpm, but for example if we added an RC filter to phase voltage signals we would have to compensate for the phase lag in the firmware.

Just a heads up.

blezalex, 2020-06-14:
I agree with you that angle error is bad, but not as bad as reading random hall values, That error can be 180 degrees :)

Here is the video showing the effect of those errors. 
https://youtu.be/EfuCU61jyk0

You'll hear 2 kind of sounds:
1 - tire squeaking 
2 - clicks and bangs coming from the motor. 

Tire is intentionally deflated, so motor works hard to spin it. All those problems gone with additional filtering.

Hall sensors data comes only once per 60 degree of rotation, so getting 3 deg error is about 5%, most likely insignificant. My motor max erpm is around 7000, and it works reliably in sensor-less mode after 4000 erpm.

blezalex, 2020-06-14:
Btw, how did you get f=8000rpm=837.758Hz?

8000erpm / 60 sec = 133 rotations per sec. 
Each hall is switching twice per electrical rotation so i'd say it is approx 133Hz. at 8000erpm.

nitrousnrg, 2020-06-15:
woops, no you were right, I typed the incorrect unit in that sheet. 8000rpm is 133Hz, and the phase error would be about 0.48° assuming the edge detection is delayed one time constant.

brycedjohnson, 2020-07-04:
@blezalex Tried this out on my diy onewheel.  7 sample filter definitely helped.  I had reduced max motor current to 50amp because that for the most part had stopped the clicking/bad torque issues.  To test I bumped it up to 60/70/80/90amp motor current and I could still see a bit of issues at 90amp when I would pitch back to stop very quickly, but in general it helped out a ton.