vedderb/bldc #201: Feature Request: Reduce/disable regen braking when battery is full

rorosaurus, 2020-08-12:
Per discussion in the forums, it sounds like regen braking will activate even when the battery is fully charged.
https://vesc-project.com/node/1999#comment-5892

This has the safety risk of starting a battery fire, or even tripping BMS and causing complete loss of control. This feature I'm requesting might be a surprise of less-than-expected regen, but I think from a safety perspective that is much more preferable!

It sounds like people are either 1) bypassing the BMS for battery discharge (to avoid BMS trip, but overcharging batteries) or 2) only charging battery to 90% to avoid this issue. Neither of which are ideal.

I assume that adjusting the regen braking on-the-fly is possible, from a hardware perspective, since the strength of the regen braking is configurable in the VESC tool. It would be nice to restrict the regen braking if the batteries are too cold, as well.

I'm not familiar with this repository, but I'm happy to take a closer look as my parts begin to arrive. I would love to help contribute if I'm able to figure out how to get set up!

nitrousnrg, 2020-08-12:
A needed feature for sure!
Benjamin knows better, but I would start by looking at how the code linearly reduces available motor current when mosfets or motor get too hot. This is handled here: https://github.com/vedderb/bldc/blob/master/mc_interface.c#L1908

That approach would require a couple of new fields in VESC Tool so we have a regen cutoff start (for example 4.05V per cell) and a cutoff end at 4.2V.

Temperature-based regen limit would be cool, but we would need that data from the BMS and vesc doesn't handle that data yet. I don't even know if there are standard BMS protocols out there to get the battery temp.

bj97301, 2023-12-29:
Should be implemented in v6.05. Regen start and end voltage I believe