vedderb/bldc #207: Duty cycle limit causes negative motor current

jkoljo, 2020-08-24:
Hi,

so my intent was to configure Duty Cycle Current Limit start to 0.85, but I ended up having both that and Max Duty Cycle set at 0.85. My mistake. However, I think what the drive did was a bit unexpected. When the duty cycle limit was reached, the drive put -29.3A to the motors. As this is an esk8 application, the board basically threw me off pretty aggressively.

https://metr.at/r/laLFU

I am using Unity, FW5.1, 2x 6380 190kv motors, 12S, PPM input, Current Smart Reverse mode. Current limits to motor are set by the detection routine, and they are about 72A each.

I would expect the drive to just go to idle (0A) if duty cycle limit is reached, not brake actively. Additionally, maybe the VESC Tool app could have some sort of a warning if Duty Cycle Current Limit and Max Duty Cycle are set as same value or very close to the same value. Based on esk8.news forum I was not the first one to make this mistake, and this would probably save ppl from quite a few hospital visits. It is especially dangerous when you consider that the surprising behavior only happens when you are close to the maximum speed of your equipment.

Thank you, br,
Juha

jkoljo, 2020-08-25:
Now back from the hospital, I tested this on the bench and its easily reproducible. Set duty cycle limit to 85% and duty cycle current limit start to 85%. Give it some throttle and when the duty cycle reaches 85%, the motors stop aggressively. And of course this braking action is aggressive enough to throw an esk8 rider completely off the board.

Jfriesen222, 2020-08-25:
There is pretty widespread reports of oscillations caused from the changes to the duty cycle limit and I've seen several reported falls. In my testing even setting it to 90% with the max at 95% seems to remove the oscillations. 

I'd suggest a temporary fix of simply changing the default value of the motor config since the vast majority of people don't go in and change the advanced settings anyway. The only problematic thing there is if people perform a config backup/restore. Having the default behavior be the safest option is best and those unaffected by the behavior can go in and change it manually opting into the risk.

angusgraham, 2021-04-21:
It's shocking to me that this hasn't been addressed at all eight months later.
Why would the motors reverse with these settings?

vedderb, 2021-04-21:
The oscillation is fixed in FW5.2. There was a bug when setting duty cycle limit and duty cycle current limit start to the same value, which is discussed and fixed here: https://github.com/vedderb/bldc/issues/262

It is in the dev-branch and will be in the next release, I just forgot to close this issue. I should probably go through the open issues one by one and update them...

closed 2021-04-21 by vedderb