vedderb/bldc #209: hw config for debug on stm32 discovery

Teslafly, 2020-09-11:
Since the default HW60_IS_MK4/3 do not work as is on the stm32f4 discovery for debug,  It would be great to have a hardware config that is specifically for the f4 discovery. 
It would connect the virtual motor by default and hardcode several input states (voltage, temp, currents?). 
Also move the leds onto the built in led's

I basically have an incomplete version of this and will submit a pr when ready.

or is one of the older configs already designed to work on the f4discovery?

Teslafly, 2020-09-14:
turns out that the entirety of the 5.1 codebase goes into a bootloop when i try to compile it. no matter what config.

hexakopter, 2020-09-15:
I guess the problem with the bootloop is now solved when using the right gcc-arm-none-eabi version?

Teslafly, 2020-09-16:
Just checked. The mk4 fw runs fine on the discovery now.

Would still be nice to have a config that uses the discovery leds and defaults to the virtual motor though.
I have a hw config that at least does the pin mapping from when I kludged on a discovery board to a power stage.

Teslafly, 2022-05-02:
I should really clean up and make a pr for my discovery hwconfig. (if that's something we want in the main repo?)