vedderb/bldc #212: Wrong PWM switching frequency by a factor of two in FOC mode

hexakopter, 2020-09-15:
I recently noticed that in field oriented control mode the MOSFET's switching frequency is off by a factor of two when looked at the motor phase voltage on an oscilloscope. With the default 25 kHz „Switching Frequency“ setting in FOC mode the actual PWM switching frequency is only 12.5 kHz instead of 25 kHz. I was expecting half that frequency for the current control loop with „Sample in V0 and V7“ disabled. I then confirmed that the current control loop frequency really runs at the expected 12.5 kHz with toggling a pin high at the beginning of the control_current function in mcpwm_foc.c and toggle it low at the end of that function and have a look at the resulting signal on the scope. On a hardware with phase shunts and „Sample in V0 and V7“ enabled  the current control loop runs at 25 kHz as expected, but the actual MOSFETs PWM frequency is also only 12.5 kHz.

I also toggled a pin high before line 2111 of the mcpwm_foc_adc_int_handler function and back to low after line 2773 to compare the m_last_adc_isr_duration value that is available from the VESC Terminal „last_adc_duration“ command with the time I measured on the scope. https://github.com/vedderb/bldc/blob/7f4c6f41f0ad946f06bd151533538d798ede9245/mcpwm_foc.c#L2112  https://github.com/vedderb/bldc/blob/7f4c6f41f0ad946f06bd151533538d798ede9245/mcpwm_foc.c#L2773

Here are two screenshots of the oscilloscope. The first one with the default 25 kHz „Switching Frequency“ setting, the second one with 60 kHz, both with „Sample in V0 and V7“ disabled.

[image]

For both screenshots:

- Channel 1 (yellow) shows the phase 1 voltage of the motor.
- Channel 2 (turquoise) shows the current control loop start and end like described above.
- Channel 3 (purple) shows the „m_last_adc_isr_duration“ like described above.

[image]

In the first screenshot the period of around 80 µs is doubled compared to the expected 40 µs for the MOSFET switching period, while the current control loop checks out at 12.5 kHz. In the second screenshot the PWM frequency is 30 kHz and not the expected 60 kHz.

While the last_adc_duration time from the VESC terminal is shown as 22 µs in both the 25 kHz and 60 kHz case it is measured as 63 µs (25 kHz) and 31 µs (60 kHz) on the scope. I have no idea why the terminal measurements are the same 22 µs in both cases while the scope measurements are different.

Because some small inductance motors really benefit from high switching frequency and people expecting the VESC foc_f_sw setting to be accurate that mistake should be fixed. I guess there are some changes in the mcpwm_foc_init function needed, but maybe someone with more programming experience has an idea how to get it corrected.
Simulations like @vedderb is showing in this great video [https://www.youtube.com/watch?v=x6lPdI9OVQg](url) differ from the VESC's real world behavior when it is only switching at half the speed. 
The explored issue also explains why there is still an audible noise with the VESC's "Switching Frequency" set in the 20-30 kHz range.

The FOC_CONTROL_LOOP_FREQ_DIVIDER in conf_general.h that should be used to get a higher pwm frequency is not working for me. When setting it to anything else than 1 the motor is just twitching and not running anymore.
https://github.com/vedderb/bldc/blob/7f4c6f41f0ad946f06bd151533538d798ede9245/conf_general.h#L281-L289

hexakopter, 2020-09-24:
I think that TODO was added a bit later by @vedderb and I was hoping that it was tested before. I had a look now and @nitrousnrg implemented the FOC_CONTROL_LOOP_FREQ_DIVIDER in this commit 9652231edb749e325adce46d213884491c348cca.
Is it possible to let us know if you tested the functionality back then @nitrousnrg? As stated above it is not working for me.

nitrousnrg, 2020-09-25:
That code was tested and the pwm scoped, it just skips the FOC control loop once in a while so you can configure higher PWM frequency without saturating the cpu. I think I needed to set 100kHz in vesc tool for a particular application.
The comment was added by benjamin, maybe he saw something I didnt.
I think I can test this #define next week, its not uncommon to see code not frequently used breaking over time, same thing happened to the virtual motor, which is the reason why I wanted to put in place automatic code coverage and unit testing, but got halfway through that, and I could only contribute CI builds.

TechAUmNu, 2020-09-26:
Just tested on my scope, DC mode is correct. FOC mode, has the pwm output always at half the set frequency in vesc tool. Changing v0v7 doesn't make a difference

nitrousnrg, 2020-09-26:
By the way, what I said I would test is the FOC_CONTROL_LOOP_FREQ_DIVIDER code.

The problem of the actual pwm switching frequency being half what s configured in vesc tool FOC->Advanced->Switching Frequency is a known and old issue for me, for a long time I've been doubling that number to get the actual frequency I want. As stated above, its super easy to tell because you can hear the 15kHz switching when you configure an impossible to hear 30kHz.

So I agree, "Switching Frequency" field should at least be renamed to "Control Loop Frequency" and somehow show the user what the resulting switching frequency will be, this shouldn't break old configs.
Personally, I would prefer to first select the actual pwm frequency and then be informed if the cpu can handle such workload or not, but I don't know how to deploy this change without breaking everyone's config.

TechAUmNu, 2020-09-26:
Ok, its caused by this line:
https://github.com/vedderb/bldc/blob/b460fa0fa3cb507a14aef26082d782f76b30d7db/mcpwm_foc.c#L347

Basically the PWM only updates after and up AND down count. Which means the PWM output is half the frequency set by the period. 

Quotes from the reference manual:

PWM mode 1
In upcounting, channel is active as long as TIMx_CNT < TIMx_CCR1 else inactive.
In downcounting, channel is inactive as long as TIMx_CNT > TIMx_CCR1 else active.

PWM mode 2
In upcounting, channel is inactive as long as TIMx_CNT < TIMx_CCR1 else active.
In downcounting, channel is active as long as TIMx_CNT > TIMx_CCR1 else inactive.

and

Center-aligned mode 1
The counter counts up and down alternatively.
Output compare interrupt flags of channels configured in output (CCxS=00 in TIMx_CCMRx register) are set only when the counter is counting down.

Center-aligned mode 2
The counter counts up and down alternatively.
Output compare interrupt flags of channels configured in output (CCxS=00 in TIMx_CCMRx register) are set only when the counter is counting up.

Center-aligned mode 3
The counter counts up and down alternatively.
Output compare interrupt flags of channels configured in output (CCxS=00 in TIMx_CCMRx register) are set both when the counter is counting up or down

trosenband, 2020-09-26:
Maybe the term "Switching Frequency" has some ambiguity.  A 10 kHz PWM waveform switches 20k times per second...

TechAUmNu, 2020-09-27:
`dt /= (float)FOC_CONTROL_LOOP_FREQ_DIVIDER; `

So for FOC_CONTROL_LOOP_FREQ_DIVIDER = 3
Pretty sure this is supposed to be *3=, not /3. You are running 3x slower for the control loop, so its going to be 3* the time.

Tested at /3 motor ramping is completely wrong and doesn't work at all. 

Changed it to 
`dt *= (float)FOC_CONTROL_LOOP_FREQ_DIVIDER; `
with *3 it works ok.

vedderb, 2020-10-10:
The switching frequency refers to how fast the modulation jumps between the 0 vectors when doing Space Vector Modulation (SVM). This means that when you scope one of the phases against ground, all three motor phases will be connected together both when the phase is a 0V and when it is at the bus voltage, which effectively halves the ripple current compared to aligning the PWM channels in the beginning. The purpose of SVM is to get the same effective switching frequency with half the amount of switching. This is not entirely true as it is not fully symmetric, which is why you get audible sound at half the set frequency, but I still think (at least back then thought) that setting the rate at which the output alters between the 0 vectors is most correct (as that is what affects the ripple current).

The control loop only runs in one of the 0 vectors, as there is no updated current information in the other one unless the hardware uses phase shunts. With phase shunts the control can actually run and update the timer in both 0 vectors, which is what the Sample in V0 and V7 option does.

vedderb, 2020-10-10:
A note about the Sample in V0 and V7 option: it will not have any effect on hardware with low side shunts as there is no new current information available when all high-side transistors are on - therefore it is ignored unless the hardware has phase shunts.

closed 2021-01-20 by vedderb