vedderb/bldc #213: Please Gyus Help Me, I Need to download Firmware Version 4.01

sahilsalman04, 2020-09-16:
Hi Guys, 
I need to download BLDC Firmware version 4.01, I am not sure how to download that.
When I flashed available Firmware from [this link](https://github.com/vedderb/bldc/) as custom firmware using VESC tool, my VESC6 MKIV hardware from trnmpn started to blink green and red led (green after that red) while blue led is continiously on. It is not getting detected after that in Linux 18.04.

It would be helpful if someone knows how to solve this issue.
1. How to get specific version(4.01) of BLDC
2. What is wrong with my VESC6 and how to solve that

Thankyou

Mitchlol, 2020-09-16:
https://forum.esk8.news/t/precompiled-vesc-tool-archive/15920

hexakopter, 2020-09-16:
Why do you need firmware version 4.01 for it? Actually the Trampa VESC6 MK4 was not supported back in that firmware and was added from Benjamin in the https://github.com/vedderb/bldc/commit/8b6f0680e496069b1b089ad55023176494bfd9af commit on 24. April 2020. 
When you have trouble with the firmware that came delivered with the ESC you bought from Trampa you should contact their customer support about that.

sahilsalman04, 2020-09-17:
Actually there is a requirement for 4.01 version, that's why i need that.
I am not sure about if i blown my device or not, i hope it's not blown.
i am thinking about flashing a bootloader into that, is there any way how to check if it's blown or not??

sahilsalman04, 2020-09-17:
Actually there is a requirement for 4.01 version, that's why i need that.
I am not sure about if i blown my device or not, i hope it's not blown.
i am thinking about flashing a bootloader into that, is there any way how to check if it's blown or not as i am just a noob with this??

sahilsalman04, 2020-09-21:
Ok Finally today i flashed Firmware in my VESC6 MKiv, it's working now.
THankyou everyone :)

Peemouse, 2020-09-21:
@sahilsalman04 May you close your issue ?

sahilsalman04, 2020-09-21:
[Peemouse](https://github.com/Peemouse), I already did that Sir..

closed 2020-09-21 by sahilsalman04