vedderb/bldc #217: After Flashing VESC Firmware, Hardware is getting bricked everytime.

sam-rke, 2020-09-29:
Hi there,
I downloaded BLDC firmware from https://github.com/vedderb/bldc , built it using terminal in UBUNTU 18.04.1 LTS and flashed bldc firmware into VESC6 MK4 using VESC tool 2.06. The SHA for above firmware is "7f4c6f4". There was no changes made in Firmware Code at first.

Now my question is when i am opening VESC tool its showing supported firmware version 5.0 and 5.1 while in Firmware "change log" the latest version is showing 5.01 i attached the image of both. I tried uploading Bootloader using ST Link V2 and ST Utility but after flashing bootloader for some 4-5 times my VESC STM Processor is damaged.

Is firmware version diffrence (i.e 5.1 in VESC tool and 5.01 written in Change log) the main cause of why my vesc was getting bricked every time when i am flashing code into hardware or anything else that i am missing.
Kindly let me know about any mistake that i am making, any help would be appreciated.
Thankyou

[image]

[image]

trosenband, 2020-09-29:
From your screenshot, it looks like you are loading default FW for HW 4.8.  Maybe try "60" for Hardware Version if you have VESC 6?  Anyway, select "Custom File" if you compiled the FW yourself.

supercrazysam, 2020-09-29:
I addressed the instructions to flash firmware in this PR
https://github.com/vedderb/bldc/pull/176

The guide to update is here:
https://github.com/supercrazysam/bldc/tree/supercrazysam-readme-update-patch

Remember to select your board in [conf_general.h]
Hope this will help you!

sam-rke, 2020-09-29:
> From your screenshot, it looks like you are loading default FW for HW 4.8. Maybe try "60_MK4" for Hardware Version if you have VESC 6 MK4? Anyway, select "Custom File" if you compiled the FW yourself.

Hi @trosenband  thanks for your response...
I am using HW VESC6 MK4 and i changed to "60_MK4" in config file as well, i flashed via custom file tab by selecting ".bin file" from "build folder" but whenever i am flashing custom firmware build by me the hardware get bricked.

sam-rke, 2020-09-29:
> I addressed the instructions to flash firmware in this PR
> #176
> 
> The guide to update is here:
> https://github.com/supercrazysam/bldc/tree/supercrazysam-readme-update-patch
> 
> Remember to select your board in [conf_general.h]
> Hope this will help you!

@supercrazysam  thanks for your update, i was using method 2 to flash custom firmware into VESC although my bin fine in build folder was having different name, maybe this is because different folder name.

But i need to confirm one thing when my VESC got bricked I used St link v2 programmer and st link utility tool over SWD pins to flash the default firmware from VESC project using [this link](https://gist.github.com/Lauszus/f7afec1f66a1ad1c8b8a0f9f35ffab69) .
So i was just flashing firmware not bootloader and VESC was working fine...but after some flashing my STM processor wasn't working (told by trampa board support as there is only continuous Blue LED ON nothing else) is there anything wrong in my process

sam-rke, 2020-09-29:
Can someone confirm is Supported Version in VESC tool and Version from changelog file is same or different in red box in both screenshots??

sam-rke, 2020-09-29:
Can someone confirm is Supported Firmware Version (5.1) in VESC tool and Firmware Version (5.01) from changelog file is same or different in red box in both screenshots??
I mean i need to clear that .1 and .01 after isn't actual reason behind bricked VESC.
@vedderb  sir if you can help into this, would be so kind of you.

trosenband, 2020-09-29:
@sam-rke: I have used the same 5.01 FW built on Ubuntu 18.04.5 from the same bldc git commit 7f4c6f on VESC6 MK4 hardware (no issues).  VESC Tool shows the FW version as 5.1, because the preceding zero in the minor version number gets dropped.

sam-rke, 2020-09-30:
> @sam-rke: I have used the same 5.01 FW built on Ubuntu 18.04.5 from the same bldc git commit 7f4c6f on VESC6 MK4 hardware (no issues). VESC Tool shows the FW version as 5.1, because the preceding zero in the minor version number gets dropped.

Okay...
thanks for confirmation @trosenband 

I have to look into it why my hardware is getting bricked everytime

sam-rke, 2020-10-03:
> @sam-rke: I have used the same 5.01 FW built on Ubuntu 18.04.5 from the same bldc git commit 7f4c6f on VESC6 MK4 hardware (no issues). VESC Tool shows the FW version as 5.1, because the preceding zero in the minor version number gets dropped.

Hi there,
I used the same procure with same FW SHA id, same VESC tool version but i don't know why my VESC is getting bricked every time when i am flashing built fw into it.
No idea if it's a hardware issue or I am doing something wrong.
BTW just to inform, I am using a "12 VOLT 1 AMP" power adaptor to power VESC.
@trosenband  or anyone else do you guys have any idea regarding possible problem?

Thankyou.
:-)

anttok, 2020-10-08:
I've been having same issue. Built using chibiStudio/Eclipse on windows, config set as follows with the rest commented out. Uploading as custom file in vesc_tool. I get a bricked vesc, and no show on com ports. Can flash back the bootloader with st link, then use vesc tool to upload the default firmware. 
 
#define HW_SOURCE "hw_410.c" // Also for 4.11 and 4.12
#define HW_HEADER "hw_410.h" // Also for 4.11 and 4.12

sam-rke, 2020-10-08:
> I've been having same issue. Built using chibiStudio/Eclipse on windows, config set as follows with the rest commented out. Uploading as custom file in vesc_tool. I get a bricked vesc, and no show [...]

Did you resolved that.
If yes how?

TomFraser, 2021-09-21:
I found compiling with any version of the Arm Embedded Toolchain other than the stated version in the README would result in a bin file that would brick the VESC. I switched to `GNU Arm Embedded Toolchain: 7-2018-q2-update June 27, 2018` as stated in the docs and can now flash via VESC_Tool or ST-Link V2 flawlessly.