vedderb/bldc #226: How to Debug VESC6 with St-Link V2

sam-rke, 2020-10-24:
Hi Guys,
I am trying to debug VESC6 firmware and I have St-Link v2 debugger.
Do someone have any idea regarding how to do Firmware debugging as I need to know source code flow.
I mean used Software and tutorial if you have any.

Thankyou

Teslafly, 2020-10-27:
I use vscode with gdb and opeoocd. 
openocd needs to be native to the os, but gdb/gcc can be installed on wsl on windows.
the xpack project maintains good distributions of openocd.

also, see this:
https://vesc-project.com/node/310

you can get gcc/gdb from here: https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads/7-2018-q2-update
not all versions will work. the latest that people seem to have success with is 
GNU Arm Embedded Toolchain: 7-2018-q2-update June 27, 2018
on ubuntu 20.04

look up any tutorial on developing stm32f4 on vscode or eclipse. the stm32f4discovery is the primary dev board that is compatible with the vesc fw

sam-rke, 2020-10-27:
> I use vscode with gdb and opeoocd.
> openocd needs to be native to the os, but gdb/gcc can be installed on wsl on windows.
> the xpack project maintains good distributions of openocd.
>
> also, see this: [...]

Thankyou for your time @Teslafly , I'll look into this, follow instructions and let you know here, if something important is there, might be helpful for someone needy like me.. :-)

sam-rke, 2020-10-27:
> I use vscode with gdb and opeoocd.
> openocd needs to be native to the os, but gdb/gcc can be installed on wsl on windows.
> the xpack project maintains good distributions of openocd.
>
> also, see this: [...]

Thankyou for your time @Teslafly , I'll look into this, follow instructions and let you know here, if something important is there, might be helpful for someone needy like me.. :)

Teslafly, 2020-10-27:
yeah, i will agree that the documentation could be better. i am in the middle of making updated guides for windows centric dev environments, but other things have gotten in the way of validating and cleaning up the guide.