vedderb/bldc #230: An approach for VESC, need you openion guys

sam-rke, 2020-11-02:
Hi there,

I am working with VESC6 and i am trying to implement some modification with which the VESC will be able to send data like motor current, its rpm and voltage to PX4 flight controller every 2 minutes.
so here my approach is...

i am trying to implement a thread in main.c, this thread will call function comm_can_transmit_sid in comm_can.c and the code will be something like in next screen shot..

[image]

Right now i dont have VESC so i am modifying the code and when this will arrive i'll test the code into it.
So my question is will that approach work??

RicardoM17, 2020-11-13:
@sam-rke if you run VESC with UAVCAN you already have this working. it won't send data every 2 minutes either but live and at high frequency.

sam-rke, 2020-11-13:
Hi @RicardoM17   thanks for your time.
I thought I have to implement a thread in VESC to send data like current, I'm also wondering if it's sending live data which files having code to inspect data like rpm, current, voltage etc from VESC to PX4?
For that I created a thread in application folder as a custom app, I'm waiting for my VESC to clear my doubts.
I'll also check based on what you suggested.

RicardoM17, 2020-11-13:
You don't have to code anything and you will be able to fly and get data via UAVCAN on the Pixhawk. and this will get logged already.

Just follow this link https://docs.px4.io/master/en/peripherals/uavcan_escs.html#vesc-esc-setup

If you want to send current commands to the VESC via UAVCAN I'm trying to work that out as well but no luck thus far. I raised the issue in #234