vedderb/bldc #233: I can't install gcc-arm-embedded with Ubuntu 20.04

albertomercurio, 2020-11-13:
Hello,
when i write `sudo add-apt-repository ppa:team-gcc-arm-embedded/ppa` in the terminal it says that there is not a version  for Ubuntu 20.04. I read that the launchpad repository is deprecated and that the latest version can be installed manually.
Do you know how to install it manually?

Jfriesen222, 2021-01-14:
Not exactly an answer but you could try simply downloading/using [chibi studio](https://www.chibios.org/dokuwiki/doku.php?id=chibios:products:chibistudio:start) if you just want to get building/testing.

vedderb, 2021-01-20:
You can download it from here
https://developer.arm.com/tools-and-software/open-source-software/developer-tools/gnu-toolchain/gnu-rm/downloads
and unpack it in /usr/local. Make sure to use the gcc-arm-none-eabi-7-2018-q2 as there seems to be a problem with gcc-versions later than 7 (I'm looking into that soon).

Quick (untested) copy-paste instructions:
1. Unpack the archive in the file manager by right-clicking on it and extract here
2. From a terminal, run
```
cp -RT gcc-arm-none-eabi-7-2018-q2 /usr/local
```

gsus24, 2023-08-17:
Faced the same problem with fresh install on 22.04.
Here is a good instruction on manually install it. See the answer from Ryall
[https://askubuntu.com/questions/1243252/how-to-install-arm-none-eabi-gdb-on-ubuntu-20-04-lts-focal-fossa](url)