vedderb/bldc #240: Please, set a milestone for release of 5.02 version

Peemouse, 2020-12-06:
Hi,

It has been a while since the work has begun on the FW5.02.
A lot of features have been added. Among them, some bug and safety fixes.

Could we set a date as a milestone for the release of this work ?
Otherwise we can endlessly wait for others features and more fixes to be added.

But no user will benefit from them unfortunately.

Thanks,

Clément

nitrousnrg, 2020-12-06:
I'm curious as well if there is a planned release date.
I'm putting together a pedal assist (PAS) pull request which is a bit involved and its slightly stressing to not know if I'll miss the next merge with master.
I have another major PR but it will most likely miss the next release anyway.

vedderb, 2020-12-07:
I agree, it would be great to get the next version out. The main thing holding me back this long is that I wanted to finish the first version of the VESC BMS firmware, as well as the BMS integration with the motor control firmware and VESC Tool. That is now mostly done, and the BMS firmware is available here:
https://github.com/vedderb/vesc_bms_fw
I have been using it for a couple of months already on some hardware under development, and it seems to work well enough for a first version.

The other thing holding me back is that I wanted to fix this problem:
https://github.com/vedderb/bldc/pull/220
I have known about it for a long time, but never had a way to reproduce it myself. Jeff is visiting me now, and he brought a motor that seems to cause it. Yesterday we spent a few hours trying to fix it, and I think we got it sorted; it would be good to get some feedback and testing on that. If that is working well now, I think we can set a date for the release.

vedderb, 2020-12-30:
I have done some more testing on https://github.com/vedderb/bldc/pull/220 and got some feedback from others, and it seems to be fixed (or at least far better) now.

For now I have set a preliminary release date to 2021-01-10. Before that I won't add new features or fixes to the current dev branch unless something urgent that needs fixing comes up.

As before, the latest build of vesc tool and the new firmware can be downloaded here:
https://vesc-project.com/node/1724

Please let me know if everything works well so far, or if there is something urgent that I should add to this release.

Peemouse, 2020-12-30:
Perfect !
I forward this news to the eskate community.

nitrousnrg, 2020-12-30:
Awesome!!

closed 2021-01-14 by vedderb