vedderb/bldc #253: ROS implementations for built in IMU ?

SAA1995, 2021-01-20:
Hi, 
I'm from University of Michigan, I have a VESC 6 MKiv and im using it for my autonomous car research project. My current implementation is similar to the F1Tenth and MIT-racecar repositories and uses ROS communication, but these do not use the built in IMU data and instead rely on servo and motor data from VESC. 

Therefore, I was wondering if such an implementation exists for IMU as well? If not could I get some intro into what would be needed to build such a package, since I am interested in using the IMU in the VESC instead of another external IMU.

Thanks.

mmoerdijk, 2021-01-20:
Hi Ahmed,

Yes there are ROS implementations, I know because I made a experimental
one. Will ask if I can share it.

Basically you will have to poll query the Vesc to send you the imu data and
then decode it on the ROS side. It is a bit different from the status
messages because the data is send in multiple can packages and the ID of
packages is setup differently (It does not use the ID of the vesc in the
can package).

The source code of Vesc_tool will tell you the details 😊

vedderb, 2021-01-20:
It is quite easy to add  more simple CAN-messages with IMU-data, and probably quite useful as the IMU-implementation in the VESC is getting better and better. It would also make sense as some applications need IMU-data at a higher rate, and then it is nice not having to poll it and crowd the can-bus with the poll messages too. Having dedicated packets also makes it possible to sync sending the packets with when the IMU is sampled, which minimizes latency.

I would suggest four messages, where each can be configured to be sent at a multiple of when the samples are taken (e.g. IMU sample rate of 200 Hz and divider 20 gives IMU packets at 10 Hz).  The four messages I suggest are:

1: Raw accelerometer data
2: Raw gyro data
3: Euler angles
4: Quaternions

They would have one divider each that can be configured, so that you can get any combination of possibles rates. What do you think?

mmoerdijk, 2021-01-20:
I think that adding separate messages for the IMU data that do not need polling is a great idea. This would it also make it easier to read the IMU data from multiple VESC on the can bus.

The messages that you suggest look good to me.

I would like to suggest adding a way to disable certain IMU messages. Raw gyro data at 10Hz is not really valuable. Maybe putting the divider to 0 should disable that can message.

SAA1995, 2021-01-20:
Hi everyone,

Thanks a lot for the information. 
@mmoerdijk if you are allowed to share, it would be a good starting point for me, but i completely understand if you can't.

I had a look at the implementations of vesc in ROS, where packets of data for motor and servo are used as you both point out, like for eg in, 
https://github.com/mit-racecar/vesc/tree/master/vesc_driver

So, im a little confused on how to start.
So, what would you suggest is a good starting point? Should I just edit the packet implementation in the repository i linked and add IMU also there OR would you suggest I start with the vesc_tool IMU code and work to make it ROS executable ? I know what I've written is super vague but i just wanted to know what from your experience you would do in this situation.

Thanks.

anscipione, 2021-05-19:
Hello,
I am creating a robot that uses 4 vescs connected to each other via can-bus and controlled throw USB with COMM_FORWARD_CAN, I would need that in the COMM_GET_IMU_DATA command can contain the controller_id as for COMM_GET_VALUES.

With this change i could know the data from which controller it comes from. I need to merge the 4 sensors to have a better localization.

in detail, adding these lines in command.c  (case COMM_GET_IMU_DATA ) makes it possible to read the IMU data knowing from which source they came :

       if (mask &amp; ((uint32_t)1 << 16)) {
      
      
        			uint8_t current_controller_id = app_get_configuration()-&gt;controller_id;
      
      
        #ifdef HW_HAS_DUAL_MOTORS
      
      
        			if (mc_interface_get_motor_thread() == 2) {
      
      
        				current_controller_id = utils_second_motor_id();
      
      
        			}
      
      
        #endif
       send_buffer[ind++] = current_controller_id;
      }

thx

itsMayankKhandelwal, 2024-02-29:
> Hello, I am creating a robot that uses 4 vescs connected to each other via can-bus and controlled throw USB with COMM_FORWARD_CAN, I would need that in the COMM_GET_IMU_DATA command can contain the controller_id as for COMM_GET_VALUES.
> 
> With this change i could know the data from which controller it comes from. I need to merge the 4 sensors to have a better localization.
> 
> in detail, adding these lines in command.c (case COMM_GET_IMU_DATA ) makes it possible to read the IMU data knowing from which source they came :
> 
> ```
>    if (mask &  ((uint32_t)1 << 16)) {
>   
>   
>     			uint8_t current_controller_id = app_get_configuration()-&gt;controller_id;
>   
>   
>     #ifdef HW_HAS_DUAL_MOTORS
>   
>   
>     			if (mc_interface_get_motor_thread() == 2) {
>   
>   
>     				current_controller_id = utils_second_motor_id();
>   
>   
>     			}
>   
>   
>     #endif
>    send_buffer[ind++] = current_controller_id;
>   }
> ```
> 
> thx

Hi, Did you successfully implemented the 4 vesc's in a 4wd robot which you were working on. Any Github links to your project  would be helpful and guide me and others on implementing something similar.

mmoerdijk, 2024-03-01:
Hey made a poc you can find it here https://github.com/Karelics/karelics_vesc_can_driver