vedderb/bldc #262: Motor stutter when braking to a stop after triggering duty cycle current limit bug

rusins, 2021-02-27:
Vesc tool 3.0.0
FW 5.2
Trampa Vesc 6+

Steps to reproduce:
(Under Motor Settings -> General -> Advanced)
1. Make sure Maximum Duty Cycle is at 95%
2. Set Duty Cycle Current Limit Start to 95%
3. Write motor configuration
4. Push remote's throttle forward, have motor spin up to max speed
5. When the motor reaches 95% duty cycle, it will suddenly freeze (Bad, I had hoped this bug had gotten fixed in FW5.2)
6. After this you can even set the Duty Cycle Current Limit Start back to 85% or 100%, write motor config, doesnt' matter – unless you restart the ESC the following occurs:
7. Spin the motor up, and now hold down brake on the remote. Instead of simply coming to a stop, the motor will stop, make loud sounds, possibly vibrate a little / stutter. It goes away if you roll the motor a bit (so that it isn't stationary), or release throttle. Braking from standstill doesn't cause this. It's only when braking while the motor was still spinning that triggers this.
8. Reboot (disconnect from power and reconnect) the vesc. Now this stuttering behavior is gone again.

Note:
Although it might be that fixing the bug in step 5. is all that is needed, this behavior worries me because I had experienced similar brake stuttering behavior when trying to set up a direct drive skateboard with another vesc. So my hypothesis is that something about this duty cycle limit bug triggers something that then allows this brake stuttering bug to be possible. I think it is really odd that even after changing settings this bug persists, and only rebooting fixes it again.

Let me know if you need a video / if there is some debug info I can collect somehow :)

rusins, 2021-02-27:
Sidenote: I was not able to trigger this bug by having duty cycle current limit start set to 94% or 96%

rusins, 2021-03-25:
I saw FW 5.3 changenotes include a fix for "motor screaming", so I tested it just now, but unfortunately I can still reproduce this bug as before. Now the motor is **really** screaming!

vedderb, 2021-03-26:
I haven't seen anything like this before. Sounds like either the motor parameters are really wrong or that the hardware is damaged.

Did you run detection?

Did you perhaps overwrite the detection values with default ones before reading them when updating that parameter?

Which remote and control mode?

What kind of motor?

What is the input voltage? Does if come from a battery or a power supply?

rusins, 2021-03-26:
Yes, I ran motor detection.

I don't believe I overwrote detection values, since they get read after the motor setup wizzard automatically. (I will check this explicitly once more, just to be sure)

Using a PWM remote (mini trigger remote), and control mode is current with brake.

Motor is a 75kv direct drive motor from Torqueboards, 28 poles. (Size is something like 68 x 100 )

Input voltage is coming from a 12s Li-Ion battery pack.

I will swap out for another vesc and motor later today just to be 100% sure it isn't HW related.

rusins, 2021-03-26:
Tried another vesc and another motor – same result on FW 5.2. Filmed a video going through motor detection, and reproducing the bug + went through some of the real-time data, in case it helps:
https://youtu.be/dlt2yOsTMUw

vedderb, 2021-03-26:
Looks to me like something is broken. Has the BMS ever cut the power while braking hard? If that has happened with both VESCs connected it is likely that both of them have died from that voltage spike and only are half-working now.

A few more things to try:
* Read the default configuration before running detection.
* Read fault codes.
* Use the sampled plot and see what the voltages/currents look like.
* Increase the switching frequency to 30 kHz.
* Halve or double the observer gain.

rusins, 2021-03-26:
On these 2 ESCs I have not had the BMS cut power. That's why I set max regen to 18A in the video, the BMS triggers at 20A. (Forgot to do this once with a stormcore, but luckily it survived the cutout) As I showed in the video, restarting the ESC or just not triggering the bug in the first place (with duty cycle settings being equal) will not cause this motor stutter, and the ESCs work perfectly from what I can tell. Definitely a FW bug in my mind...

After this behavior there are no faults logged.

I will try the last few things you suggested though and see if they make a difference.

avlasic, 2021-03-26:
@vedderb 

This is caused by utils_map returning NaN in duty cycle limit when both limits and current duty are the same value. 
I observerd that with foc_state command returning NaN for target Iq, and obviously current controller doesn't like that. 

Also same thing happens when d axis gain scalling options are both at the same value.

vedderb, 2021-03-27:
You are right. I never came to trying to set them to exactly the same value, as I thought this was related to the current controller becoming unstable when running out of voltage to regulate the currents (which is a confirmed issue on some setups and the reason for the gain scaling). In my tests I set it to the highest value as I though that would be the worst case. I will look into this tomorrow and go through all places where utils_map is used.

vedderb, 2021-03-27:
This problem is actually more tricky than I thought. I was thinking of just avoiding the numerical instability, but the problem is that setting the parameters close to the instability is inherently an unstable configuration. For example, setting Duty Cycle Current Limit to 94% is asking it ramp down over a very narrow range, so when hitting full speed the current controller will oscillate on the between the set current and 0, which probably isn't good. Same think with the d axis scaling - setting them too close is configuring a heavy oscillation in the gain when approaching that duty cycle.

I think I will avoid the instability with some trick and also check the parameters so that an unstable configuration is not requested.

vedderb, 2021-03-27:
I went through all uses of utils_map and tried to avoid the problem with appropriate margins based on each case:
https://github.com/vedderb/bldc/commit/2133d21fbc4b3a33d970273b5382ec0ae306978f
The behavior of Duty Cycle Current Limit Start is also changed slightly to mean a percentage of the limit itself instead of an absolute limit, as I think that is more natural.

The test builds are updated now too:
https://vesc-project.com/node/2859

Can you give the latest version a try and see if it solves the problem? I was not able to reproduce the problem myself for some reason, even when setting Duty Cycle Current Limit Start to the same as Maximum Duty Cycle before the fix.

rusins, 2021-03-27:
> The behavior of Duty Cycle Current Limit Start is also changed slightly to mean a percentage of the limit itself instead of an absolute limit, as I think that is more natural.

I fully agree! :)

> I was not able to reproduce the problem myself for some reason, even when setting Duty Cycle Current Limit Start to the same as Maximum Duty Cycle before the fix.

Oh wow, strange!

> Can you give the latest version a try and see if it solves the problem?

Tried and I can no longer trigger this issue! :partying_face:  Tried current limit start values 85%, 95%, 99%, 100%; can't get the motor to lock up, so braking remains fine.

Small sidenote:
Setting max duty cycle current limit start to 99% (or anything close to 100%) makes the motor make sounds that sound like electricity arcing on the bench at 95% duty cycle, but I think that's always been the case / not an issue when under load. Probably has something to do with 1% being too small of a gap to accurately increase or decrease current.

Thank you for fixing this! :)

vedderb, 2021-03-27:
Perfect! I'm closing this issue then. Thanks for reporting and helping with the testing!

closed 2021-03-27 by vedderb

rusins, 2021-03-28:
One thing I just remembered is that if duty cycle current limit start now means a percentage rather than an absolute value, it would make sense to change the default 85% (for electric skateboards) to 90%, so that it behaves similar to before.

vedderb, 2021-03-31:
85% is just some value I came up with, I have no idea what most people prefer. The lower this value is, the longer the torque ramp down is as you are approaching full speed (and the more top speed it costs you). If you do some testing and find a good default value I can use that in vesc tool for e-skate. Personally I set my speed limit much lower than top speed, so I never reach speeds where I would notice any difference.

muhammadyanuar230104, 2025-02-20:
hai, how can i get the file for vesc version 3.0.0