vedderb/bldc #265: No braking when over MCCONF_L_RPM_MAX

sullin, 2021-03-11:
Once the motor speed goes over MCCONF_L_RPM_MAX, the motor current is limited to zero - for both acceleration and braking. This causes a dangerous issue running downhill as all control is lost.

vedderb, 2021-03-12:
I tested this now on my motor bench with one motor driving another with a higher speed than the configured speed limits. The test was made in both directions with and without inverted motor direction.

* VESC Tool control: everything ok
* VESC Wand control: everything ok with all combinations of modes and directions
* PPM control: one problematic combination found: Current Hyst reverse with brake. When turning the motor backwards, accelerating forwards was not working. I found the problem and fixed it for the next release.

What firmware version are you on, what control mode are you using and how are you making the test?

sullin, 2021-03-12:
V5.1. Our own high-level control, using mc_interface_set_pid_speed().
This happened also in real-world (robotics), but in lab I have 2 motors back to back. One gets driven using mc_interface_set_pid_speed() and lower L_RPM_MAX, the other is driven with mc_interface_set_current and higher L_RPM_MAX. When raising the speed of the first motor to the max, the other motor pulls the first one to its limit and the first one becomes uncontrollable.

closed 2021-03-13 by vedderb