vedderb/bldc #266: SPI Trouble

CTSchorsch, 2021-03-16:
Hi,

i want to use the SPI interface to communicate with a RFM95W LoRa module. I have some problems to get any action on the mcu pins :(. Maybe it is just one step 

what i did:
I want to use SPI3. The pins are available on my hardware

```
#ifdef HW_HAS_RFM95W
#define HW_RFM95W_SPI_DEV               SPID3
#define HW_RFM95W_SPI_GPIO_AF           GPIO_AF_SPI3
#define HW_RFM95W_SPI_PORT_NSS          GPIOA
#define HW_RFM95W_SPI_PIN_NSS           15
#define HW_RFM95W_SPI_PORT_SCK          GPIOC
#define HW_RFM95W_SPI_PIN_SCK           10
#define HW_RFM95W_SPI_PORT_MOSI         GPIOC
#define HW_RFM95W_SPI_PIN_MOSI          12
#define HW_RFM95W_SPI_PORT_MISO         GPIOC
#define HW_RFM95W_SPI_PIN_MISO          11
#define HW_RFM95W_SPI_PORT_DI0          GPIOC
#define HW_RFM95W_SPI_PIN_DI0           5
#endif
```

create a hwconf/rfm95w.c and add a init function to configure the pins and start SPI driver
```
 palSetPadMode(HW_RFM95W_SPI_PORT_SCK, HW_RFM95W_SPI_PIN_SCK, HW_RFM95W_SPI_GPIO_AF | PAL_STM32_OSPEED_HIGHEST);
 palSetPadMode(HW_RFM95W_SPI_PORT_MISO, HW_RFM95W_SPI_PIN_MISO, HW_RFM95W_SPI_GPIO_AF | PAL_STM32_OSPEED_HIGHEST);
 palSetPadMode(HW_RFM95W_SPI_PORT_NSS, HW_RFM95W_SPI_PIN_NSS, PAL_MODE_OUTPUT_PUSHPULL | PAL_STM32_OSPEED_HIGHEST);
 palSetPadMode(HW_RFM95W_SPI_PORT_MOSI, HW_RFM95W_SPI_PIN_MOSI, HW_RFM95W_SPI_GPIO_AF | PAL_STM32_OSPEED_HIGHEST);
 palSetPadMode(HW_RFM95W_SPI_PORT_DI0, HW_RFM95W_SPI_PIN_DI0, PAL_MODE_OUTPUT_PUSHPULL | PAL_STM32_OSPEED_HIGHEST);
 spiStart(&HW_RFM95W_SPI_DEV, &rfm95w_spi_cfg);

```

inside the working thread a do just a test transmit to check on the oscilloscope

```

 spiSelect(&HW_RFM95W_SPI_DEV);
 spiSend(&HW_RFM95W_SPI_DEV, 1 , 0x55);
 spiUnselect(&HW_RFM95W_SPI_DEV);

```

But nothing happens on MOSI or CLK. Can someone see a mistake ?
Thanks
Georg

vedderb, 2021-03-26:
Did you activate SPI3 in mcuconf.h? If you did it might not work anyway as I2C2 uses the same DMA stream for tx. There might be a different DMA stream available on one of them for a combination that works, you can check the table that is somewhere in the long datasheet. Alternatively, SPI can be run quite well in software as it is synchronous and timing is not critical.

CTSchorsch, 2021-03-26:
Hi Benjamin,
yes, SPI3 was already activated, but i did not take care about the DMA streams :-( 
But as you mentioned, i did SPI in software and the communication to the RFM95 board works fine. 
Now i work on the integration in your command/packet system, which is well prepared for extensions. thanks.

As I start the RFM95 thread from my hw_gesc.c worker thread i had to add a void isInitialized() to your command.h. Without that, the system was not come up. I think that the hw thread is started too early.

Maybe the LoRa module is interresting for some glider pilots to control the winch. LoRa work up to 10km in air and my first tests under water were also very promising.

vedderb, 2021-03-26:
I think a RFM95-driver that can be used by any hardware would be nice. I just ordered a few units, so I can do some testing too and help out with the driver.

CTSchorsch, 2021-03-26:
nice :) i used the this driver 
https://github.com/realspinner/SX1278_LoRa
and modified the SX1278_hw.c to do the software spi.