vedderb/bldc #274: servo_out firmware for FSESC hardware

uwmyuan, 2021-03-31:
Hi,

I'm using VESC6 compatible hardware from FlipSky for building a ROS vehicle ([hardware](https://flipsky.net/collections/new-accessories/products/high-current-fsesc-200a-60v-base-on-vesc6) comes with the firmware `VESC_default_no_hw_limits` 5.2). The VESC needs an HW60 and servo_out firmware. Since the servo_out.bin is not an option in vesc_tool, so I built the firmware by following the [instruction.](https://groups.google.com/g/f1_10/c/W6MxZ6bdzdk/m/7YtjVEw2CQAJ) with the default hardware option (i.e. vesc mkv). 

The servo works well, however, the motor does not work, and the motor does not spin during FOC. During FOC, I found the parameters were abnormal (R 520.20 mOhm, L 29.64 muH, 100.000 mWb, KP 0.0296, KI 520.20 Observer Gain 0.10). Does this matter? And I tried other hardware compiling options, and they had the same issue. 

Do you have any debugging clues? Thanks!

vedderb, 2021-03-31:
You have to ask flipsky about that. I have no way to know what their schematic looks like and which hardware file is compatible.

It would be nice to make servo_out work with the default firmware though with a config option, as many people seem to be using it. I will look into that.

closed 2021-03-31 by vedderb

vedderb, 2021-03-31:
I just made an update so that you can use the servo output with the default firmware:

[image]

Just make sure that the PPM-app is disabled. This change will be in the next firmware and vesc tool release, but if you want to test it now already there is a prebuilt version here:
https://vesc-project.com/node/2859

uwmyuan, 2021-03-31:
Fantastic! I tested it, and it works well. We appreciate your kind response.