vedderb/bldc #278: Docker images for firmware compilation

Teslafly, 2021-04-09:
This issue is for feedback on creating dockerfiles to compile vesc tool/ vesc fw
It seems that some are having some issues with firmware compilation.

https://github.com/vedderb/bldc/issues/210
https://github.com/vedderb/bldc/issues/233

I use docker extensively for firmware and library compilation at my work with good results. It's great because it isolates your build environment from your system. Allowing an identical setup every time across systems.

Would this be something worth including in the vesc repo itself to help beginners get up and running? It is possible to trigger a docker container from a make file. Having it rebuild then compile the firmware. 
If not, I will probably make my own repo to host them. vesc_tool is especially a pita to get the build environment set up.

vedderb, 2021-04-09:
I'm not that familiar with docker, but if it just requires a few text files that set up the containers it would make sense to add. If you look into it I can test it and see if it works.

Teslafly, 2021-04-09:
Yeah, docker compose is mainly just text files that set up a low overhead virtual environment.
I'll see if I can get something reasonable working this weekend.

supercrazysam, 2021-04-12:
But there is really just one "gcc-arm-embedded" installation involved.  (except for the MacOS side)

I think the problem here is that people need to know which "gcc-arm-embedded" version they should be using.

For the solution, I think it would be better to just list out the tested working version of gcc-arm-embedded, explicitly stating which version is stable for development to install. 

Introducing another bunch of dependencies and extra steps to install docker, get a docker container running, get familiar with docker is not really a minimalist & beginner friendly solution imo.

Teslafly, 2021-04-12:
yes and no. you only have to install two things, docker and openocd. docker takes care of all the rest of the dependencies. It provides a system independent way of compilation, and the (current) installation instructions do not work on ubuntu 20.04. Not to mention some people may have different versions of gcc_arm that break vesc. The older version does not work with the stm32g474 I use in a different project, so I cannot install globally. (inconvenient, but overcome-able)

I already have vesc fw compile working in docker. I just need to clean it up and tie it to a script to make it public ready

Then there is vesc tool, which qt is a royal pita to get working. I still have not been able to get it working on WSL, which has meant I haven't been able to contribute to vesc tool at all. And there are a couple things I would like to do. I wrote some custom features for a battlebot that currently has hardcoded values that really should be adjustable through vesc tool.

Teslafly, 2021-04-12:
Though we should have instructions for both.  A local install has its own merits. docker is just guaranteed to always work

vedderb, 2021-04-12:
Another approach is pointing the makefile to e.g. $HOME/gcc_arm_ebbedded/2018-q2 and have a script that downloads it there. It seems to run on any distro and kernel version, and you only need gnu make in addition from the package manager. There could also be a gcc directory in the firmware directory with a script that download it there, and the path in the makefile can be relative.

ddosoff, 2021-04-21:
I am interested in docker images for bldc firmware and vesc_tool. This will be very convenient. 

Any progress?

Teslafly, 2021-04-22:
I have the vesc fw dockerfile working in that it compiles code, but wanted to add a few utilities to get the debugger to work with vscode/eclipse as well.

dockerfile:
```
FROM gcc:7.2
ENV REFRESHED_AT 2017-11-21

RUN apt-get update \
    && apt-get -y install git 
    # scons bzr lib32z1 lib32ncurses5

# Set up a tools dev directory
WORKDIR /home/devtools

# pull the gcc-arm-none-eabi tarball
RUN wget https://developer.arm.com/-/media/Files/downloads/gnu-rm/6-2017q2/gcc-arm-none-eabi-6-2017-q2-update-linux.tar.bz2 \
    && tar xvf gcc-arm-none-eabi-6-2017-q2-update-linux.tar.bz2 \
    && rm gcc-arm-none-eabi-6-2017-q2-update-linux.tar.bz2

# Set up the compiler path
ENV PATH $PATH:/home/devtools/gcc-arm-none-eabi-6-2017-q2-update/bin

WORKDIR /usr/project
```

use with:
```
docker build -t vesc_fw_cont .
docker run --rm -v "$(pwd)":/usr/project vesc_fw_cont:local make
```

I have not made much progress on the vesc tool docker compiler. Many packages are missing from later ubuntu versions and i don't want to go to ubuntu 16 vs 18 or 20. I also haven't had a lot of time.

There are a couple docker qt compiler templates, but I have to figure out how to adapt them for the specific version/s vesc is compatible with. It doesn't help that vesc is really my only experience with qt.

The vesc_tool compiler docker is definitely the more valuable of the vesc tool vs firmware.

Teslafly, 2021-04-22:
The other thing with vesc tool is qtcreator. If you don't want to make a qt account/navigate their confusing options, which seems to be needed to do anything with the native windows installer, there are options like https://github.com/dannywillems/docker-qtcreator

tomatac, 2022-01-16:
This is a great idea! Any progress in creating a Docker file for compiling the firmware and the vesc tool?
I have been using Docker for development for few years and is excelent. You can move to a new computer and have the right development environment in minutes and makes it easy for anybody to join a development team.
Also VS Code makes it easy to compile and debug in a container.

Teslafly, 2022-05-02:
with the refactoring of the makefile, it is now fairly reliable to build the firmware on both linux and windows systems. for the firmware there is less value in this. probably still some value in this for creating vesc_tool standalone binaries easially but his issue can be closed for now.

closed 2022-05-02 by Teslafly