vedderb/bldc #290: Selecting "No App" semi-bricks my VESC (instantly disables Bluetooth)

surfdado, 2021-04-18:
In order to prevent writing firmware while the motor is running (balance app - brake current) I switched my app to to "No App". This was a big mistake. Whenever I do that I instantly lose the ability to communicate with my VESC via bluetooth. I was forced to reconnect via USB (had to take my whole build apart to access the USB port) to select a different app...

I tried it a second time to make sure that I can reproduce it and I got the same behavior.

For now, I will switch to the UART app before writing firmware, but it would be nice if "No App" wouldn't render Bluetooth unusable.

PS: My VESC is the Little FOCer

Peemouse, 2021-04-18:
Hi, this is normal. The bluetooth module uses UART port, so you need to keep it enabled for using bluetooth.

surfdado, 2021-04-19:
Haha, okay - learned something new. I guess you we can close this issue then and file it under user error ;)

closed 2021-04-19 by surfdado