vedderb/bldc #296: CAN_PACKET_PING not working correctly on 4WD setups with two dual ESCs

janpom, 2021-04-25:
When doing a CAN ping on a 4WD setup with two dual ESCs (Stormcore/Unity), only three CAN IDs are returned. The ID of the secondary motor on the slave ESC is missing. This is reproducible on FW 5.2, but not in 5.1. 

I suppose this is the change that caused the issue: 242a1924e1c5f97661579c7a43a2171671ecaf0f.

vedderb, 2021-04-27:
I think I found the problem: https://github.com/vedderb/bldc/commit/fa3610c8b165fb6206659f2902ef903c5ff9cbbb
It is that the ping command since a few versions verifies that the reply comes from the ID that the ping was sent to, and the dual hardware would always reply with the first ID. The test build is now updated:
https://vesc-project.com/node/2859
Please give it a try and let me know if it works.

janpom, 2021-04-28:
I'm unable to test. I'm on a mac, so can't use the pre-compiled VESC Tool. I built the firmware using the `dev_fw_5_03` branch but when I installed it on my Xenith, it bricked it. If you can provide the Unity .bin file, I'm happy to test it.

vedderb, 2021-04-28:
[VESC_default_unity_2021-04-28.zip](https://github.com/vedderb/bldc/files/6391027/VESC_default_unity_2021-04-28.zip)
This one is for the unity. Configuration will only work with the latest beta build of vesc tool, but you should be able to test the can scan.

janpom, 2021-04-28:
Thanks for the file. This seems to be working correctly. I tested both with the [VESC Tool](https://drive.google.com/file/d/1F5Dom3oOfXHXSPbgHMNXnWKROUZUI8Cj/view?usp=sharing) and with my [DAVEGA](https://drive.google.com/file/d/1b5S4CMFrJBA7qITJR4ZLE_VmTJ0zm_kK/view?usp=sharing). Both looks good. 👍

vedderb, 2021-04-28:
Perfect! I'm closing this then.

closed 2021-04-28 by vedderb