vedderb/bldc #304: Parameter get/set interface over CAN

imcnanie, 2021-05-31:
I am writing a CAN interface for the VESC and I've managed to implement all the realtime commands.  But when it comes to getting and setting parameters, it seems like only some of the parameters are implemented over CAN, namely current limits, foc_erpms, and battery cutoff. I'm looking at comm_can.c and datatypes.h for reference. Where are all the other commands for the App modes and motors modes? Ideally I'd like to be able to parse and send over a mot_conf.xml and app_conf.xml, and update the firmware on individual VESCs over CAN. Am I missing something?

bsilver8192, 2021-06-02:
I'm pretty sure you can do this with `comm_can_send_buffer`. I haven't used it directly, but that's how the VESC Tool CAN forwarding functionality works, and I know you can do everything including parameter and firmware updates that way.

The other CAN wrappers are all single-frame versions that (mostly) trigger the same function calls as `comm_can_send_buffer`. `comm_can_send_buffer` can't fit as much data into a single CAN frame though, so it will use more bus bandwidth and be slower to do equivalent things. The flip side, of course, is that it can access all of the behavior that the USB serial port interface can.

For example, compare the handling of [`CAN_PACKET_SET_RPM`](https://github.com/vedderb/bldc/blob/be439a2a55ae5b1ba8cce8ad6b0da2d80466e9df/comm_can.c#L1177) (which is what `comm_can_set_rpm` triggers) to [`COMM_SET_RPM`](https://github.com/vedderb/bldc/blob/be439a2a55ae5b1ba8cce8ad6b0da2d80466e9df/commands.c#L414).

A bit more detail: `comm_can_send_buffer` splits things up into data CAN frames, sends those, and then sends a `CAN_PACKET_PROCESS_SHORT_BUFFER` or `CAN_PACKET_PROCESS_RX_BUFFER`, which then calls [`commands_process_packet`](https://github.com/vedderb/bldc/blob/be439a2a55ae5b1ba8cce8ad6b0da2d80466e9df/comm_can.c#L1231), which is where the `COMM_SET_RPM` handler I linked above lives.

imcnanie, 2021-06-04:
That makes sense, I'll give it a try! As far as firmware update over CANbus, do you have any notes on that?

bsilver8192, 2021-06-05:
Unfortunately not. If you can't figure stuff out by reading the code on both ends (VESC tool and the VESC code), you could hook a CANable up and dump all the frames while you do a firmware update via CAN forwarding from VESC tool.