vedderb/bldc #305: STM32 F4 Discovery

GI101, 2021-06-05:
Hello all,

Supported Boards all of them but which number is compatible STM32F429, 180Mhz MCU with embedded ST link/v2? 

"Make sure you select your board in conf_general.h"

The STM Discovery launch pad has 2 rows of header pins and 2.4" touch screen LCD. My test inverter is now connected to TI C2000 LauchXL-x49c launch pad with SDK sensorless motor control FOC labs to drive 3 phase SPM motor.  

Will VESCA work on TM4C1294 120Mhz ARM processor with 2 ADC's 16 Chan, 8 timers, 2QEI, many GPIO's. Each ADC has fixed step sequencers 2x4, 2x8 and currently access CMISS library functions.  It perhaps can support the VESC boot loader but what about the SVM code? I only ask since the DC inverter uses Trapezoidal on custom PCB with parallel NFETS and very expensive.

Regards,

Teslafly, 2021-06-06:
This is pretty outside the scope of the vesc project. This firmware currrently) only runs on stm32f405/407. The STM32F429 is not supported that I know of.
Porting to a new mcu is hard work and not explicitly supported.

That said, the first step would be to get it running chibios, and then transitioning over the peripherals. many of which probably have different interfaces.
The STM32F429 may be possible as it is sort of close to the 407, but the TI chip is going to be a comparable amount of work to writing firmware from scratch. Especially since TI offers instaspin as a motor commutation base.

GI101, 2021-06-07:
Hello Teslafly thanks for helping.

Had figured it was an exaggeration to say "works of all boards". Hoping VESC don't use RTOS Bios (chibios?) but seemingly ST-HAL is more acceptable. I already have TM4C (ARM-4M) sensorless trapezoidal FOC firmware with excellent PI speed control. Created parts of SVM code such as Clarke alpha/beta, co/sin but not 6 sector translations needed to drive Park switching table. The invPark was not part of ARM CMSIS code & was largest part, then there is newer GDPWM reduced switching losses to consider. The TMS280049c SDK firmware runs PWMD1 pattern but I don't like phase current drive so uneven, TI support is not owning up to why that occurs or offing any ideas how to resolve it.  
[image]    

The C2000 PI speed control has issues with regulating current for large rotor inertia mass. The inertia kgm2 formula was never bench tested by TI engineers on larger motors, integral suffers badly. The C2000 has ROM embedded FAST angle estimator, low speed observer for slow speed startup <1Hz besides the community clamor around CLA module off loaded functions. [image]

What do you think about Neucleo 431 to support VESC? [ https://www.st.com/content/st_com/en/products/evaluation-tools/product-evaluation-tools/mcu-mpu-eval-tools/stm32-mcu-mpu-eval-tools/stm32-nucleo-boards/nucleo-g431rb.html](url)

GI101, 2021-06-07:
So VESC firmware loads on top of a cibios, that makes me a bit apprehensive. TI-RTOS on TM4C 120Mhz CPU ran turtle slow and had minor kernel tasking issues but had an M4 NVIC module that easily supported 12 peripheral interrupts with debug trace stacks.

Yet the ST429 180Mhz CPU has few more DMIPS  and chibios has ports drivers. Wonder if chibios may already be loaded on the F429 Discovery kit, haven't powered up over year or so? It seems as if chibios supports ST431 Nucleo launch pad too. There was also peripheral driver stm32fxx.mk in the Git file repository.  [ChibiOS_3.0.5/os/common/ports/ARMCMx/compilers/GCC/mk/startup_stm32f4xx.mk](url)  

It would seem VESC firmware needs to be matched to specific GPIO ports and ST429 driverlib should support like VESC named peripheral functions. The difference in the memory maps (407 to 429) would be another issue to consider. I haven't yet dig into the code that deep but noticed it was very linear type coding style. This TM4C FOC code is mostly all interrupt driven and has very fast current monitor PWM update loop 329µs.