vedderb/bldc #314: Multiple CAN RX buffers

rpasichnyk, 2021-07-02:
This issue will become more problematic since the number of CAN devices increases. When talking to VESC each device that sends a command that does not fit into a single payload wants to use `rx_buffer`. If those devices send at the same time, buffer will contain mixed data and nothing will work.

I create this issue to brainstorm possible solutions to this problem before working on a pull request. What can be done?

One simple idea is introducing `rx_buffer1`, `CAN_PACKET_FILL_RX_BUFFER1`, `rx_buffer2`, `CAN_PACKET_FILL_RX_BUFFER2`, etc. or maybe something more flexible like `CAN_PACKET_FILL_RX_BUFFER_N` where each frame contains buffer number?

What are your thoughts on this?

supercrazysam, 2021-08-06:
In industrial application, people use something like CANopen. 
Maybe we can take a look at how they implemented it? 
<del>(If only VESC is CANopen compatible)</del>

vedderb, 2023-01-02:
Here is an attempt at multiple buffers:
https://github.com/vedderb/bldc/commit/0dec878d1d335485ac3f313b4136c8e8bd354477
It is not foolproof, but seems to work much better so far. It is also backwards compatible, which is important.